buserror/simavr #108: Building on OS X with CrossPack-AVR and homebrew doesn't work

deoxxa, 2015-01-01:
By default, simavr wouldn't build on my laptop. I've got [CrossPack-AVR](http://www.obdev.at/products/crosspack/index.html) providing my avr toolchain, and libelf etc. installed from homebrew (into /usr/local). The makefile as it is, and one of the c files, assume some things that don't match with what I have installed.

I patched things as shown below to make simavr play nicely with my system. I hope this can help someone else later! I've also included a log of the errors I got before this patch, in case someone googles them some day.

```diff
diff --git a/Makefile.common b/Makefile.common
index 42c7004..3ea2352 100644
--- a/Makefile.common
+++ b/Makefile.common
@@ -46,13 +46,13 @@ CORE_CFLAGS	= -nostdinc -DAVR_CORE=1
 ifeq (${shell uname}, Darwin)
 # gcc 4.2 from MacOS is really not up to scratch anymore
 CC			= clang
-AVR_ROOT 	:= "/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/"
-AVR_INC 	:= ${AVR_ROOT}/avr-4/
+AVR_ROOT 	:= /usr/local/CrossPack-AVR
+AVR_INC 	:= ${AVR_ROOT}/avr/
 AVR 		:= ${AVR_ROOT}/bin/avr-
 # Thats for MacPorts libelf
-ifeq (${shell test -d /opt/local && echo Exists}, Exists)
-IPATH		+= /opt/local/include
-LFLAGS		= -L/opt/local/lib/
+ifeq (${shell test -d /usr/local && echo Exists}, Exists)
+IPATH		+= /usr/local/include
+LFLAGS		= -L/usr/local/lib/
 endif
 else
 LINUX_AVR_ROOTS := /usr/lib/avr /usr/avr /opt/cross/avr/avr /usr/local/avr
diff --git a/simavr/sim/sim_elf.c b/simavr/sim/sim_elf.c
index a0a49ee..3732ae9 100644
--- a/simavr/sim/sim_elf.c
+++ b/simavr/sim/sim_elf.c
@@ -29,8 +29,8 @@
 #include <stdlib.h>
 #include <stdio.h>
 #include <string.h>
-#include <libelf.h>
-#include <gelf.h>
+#include <libelf/libelf.h>
+#include <libelf/gelf.h>

 #include "sim_elf.h"
 #include "sim_vcd_file.h"
```

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr RELEASE=0
CONF obj-x86_64-apple-darwin13.3.0/cores.deps
WARNING cores/sim_90usb162.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1280.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1281.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1284.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128rfa1.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128rfr2.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega16.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega164.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega168.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega169.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega16m1.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega2560.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega32.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega324.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega328.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega48.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega644.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega8.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega88.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny13.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny2313.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny24.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny25.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny44.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny45.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny84.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny85.c did not compile, check your avr-gcc toolchain
/Applications/Xcode.app/Contents/Developer/usr/bin/make obj config
make[2]: Nothing to be done for `obj'.
CONF sim_core_decl.h
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
CC sim/sim_avr.c
CC sim/sim_core.c
CC sim/sim_cycle_timers.c
sim/sim_elf.c:32:10: fatal error: 'libelf.h' file not found
#include <libelf.h>
         ^
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin13.3.0/sim_elf.o] Error 1
make[1]: *** [all] Error 2
make: *** [build-simavr] Error 2
```

xfbs, 2015-02-11:
Have a look at the pull request I just made: [#114](https://github.com/buserror/simavr/pull/115). That should fix it :+1:

ft232r, 2017-01-07:
there is nothing like https://github.com/buserror/simavr/pull/115/commits/782879012107000ed0ecbf095951c6de32f6549f in master branch, have PR https://github.com/buserror/simavr/pull/115 been merged successfully?