buserror/simavr #116: board_simduino segfaults on run

thirtythreeforty, 2015-04-16:
It compiles fine, and seems to try to initialize, but it prints a bunch of garbage and segfaults before anything can be done.

Git tells me 3c5e1ae3f560ba1568da247f4e835206ff12a264 is to blame:
```
$ git bisect bad
3c5e1ae3f560ba1568da247f4e835206ff12a264 is the first bad commit
```
The backtrace at this commit is as follows:
```
#0  0x00007ffff7bcba80 in pthread_kill () from /usr/lib/libpthread.so.0
#1  0x0000000000402bb2 in uart_pty_stop (p=0x69f780 <uart_pty>) at ../parts/uart_pty.c:246
#2  0x0000000000405682 in avr_reset (avr=avr@entry=0x6a1010) at sim/sim_avr.c:132
#3  0x000000000040579b in avr_init (avr=0x6a1010) at sim/sim_avr.c:97
#4  0x0000000000401f06 in main (argc=1, argv=0x7fffffffdb18) at simduino.c:210
```
Please tell me if I can help further.  Hopefully this is something simple, although I don't know much about the structure of this project.

buserror, 2015-04-16:
Thanks for the bissect; @bsekisser any idea?

bsekisser, 2015-04-16:
While I could be wrong, I'm inclined to think it is coincidence...  looking at the backtrace, the fault is happening during setup/reset/initialization before any avr code starts running...  before the commit in question would have much influence...  unless a timer was being set up?
were you just running the simduino example?..  nothing special?
won't be able to look into it much till the weekened, but I may give a look over for anything I may be able to bring to light.

buserror, 2015-04-16:
I'll have a look too, seems bizare as I agree that it looks pre-reset anyway... Also, it's not a 'new' commit, and I'm sure someone must have run that example since...

thirtythreeforty, 2015-04-16:
Let me know if you can't reproduce it. And yeah, it struck me as odd that
nobody would have hit it before now.

bsekisser, 2015-04-16:
@busserror
Managed to do some testing here with what I got...
doing my own bisect suggests commit e6476edb26cb403eed6365601b8a33abcfe41b9d; flash: Add support for TEMPPAGE writing.

buserror, 2015-04-18:
Do you actually crash? Here it runs just fine (!) -- typical isn't it :-)

bsekisser, 2015-04-18:
I just did a push...  flash write was failing...  z was pointing to the last word in the page and write ended up writing into the next page...  not sure if I got the behavior correct but it looks to work in this instance.  the most reliable thing to do is track actual write address and use that for the flash address to write, but again don't know what the actual behavior is supposed to be and how it is handled by the avr.

closed 2016-11-07 by buserror