buserror/simavr #123: I had an issue running make install

kaayr1m, 2015-07-12:
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr install RELEASE=0
/Applications/Xcode.app/Contents/Developer/usr/bin/make obj config
make[2]: Nothing to be done for `obj'.
make[2]: Nothing to be done for `config'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
sim/sim_elf.c:32:10: fatal error: 'libelf.h' file not found
#include <libelf.h>
         ^
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin14.0.0/sim_elf.o] Error 1
make[1]: *** [all] Error 2
make: *** [install] Error 2

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr install RELEASE=0
/Applications/Xcode.app/Contents/Developer/usr/bin/make obj config
make[2]: Nothing to be done for `obj'.
make[2]: Nothing to be done for `config'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
sim/sim_elf.c:33:10: fatal error: 'gelf.h' file not found
#include <gelf.h>
         ^
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin14.0.0/sim_elf.o] Error 1
make[1]: *** [all] Error 2
make: *** [install] Error 2

kaayr1m, 2015-07-12:
I fixed this by changing the includes :

#include <libelf.h>
#include <gelf.h>

TO :
#include <libelf/libelf.h>
#include <libelf/gelf.h>

githubgilhad, 2017-01-15:
I had the same problem on **gentoo**, resolved it by installing **dev-libs/libelf**

closed 2018-04-24 by buserror