buserror/simavr #131: Compile errors on gcc 5.2.0 under arch-linux

TheCrazyT, 2015-08-19:
If i compile under arch-linux wich has the newest gcc i get the following errors:

    In file included from atmega88_uart_echo.c:22:0:
    atmega88_uart_echo.c:34:23: error: initialization of flexible array member in a nested context
      { AVR_MCU_VCD_SYMBOL("UDR0"), .what = (void*)&UDR0, },
                           ^
    ../simavr/sim/avr/avr_mcu_section.h:142:10: note: in definition of macro 'AVR_MCU_VCD_SYMBOL'
      .name = _name
          ^
    atmega88_uart_echo.c:34:23: note: (near initialization for '_mytrace[0]')
      { AVR_MCU_VCD_SYMBOL("UDR0"), .what = (void*)&UDR0, },
                           ^
    ../simavr/sim/avr/avr_mcu_section.h:142:10: note: in definition of macro 'AVR_MCU_VCD_SYMBOL'
      .name = _name

I made a Dockerfile that reproduces the exactly same problem:
https://gist.github.com/TheCrazyT/cef7b8ee796845a76bf4
(build it with "docker build -t archtest ." , run make-command with "docker run -ti archtest")

Because I can compile under ubuntu (gcc version 4.8.4) i guess it has something to do with changes in gcc.

buserror, 2015-08-20:
Wierd; is it not compiled with -std=gnu99?

TheCrazyT, 2015-08-20:
yes it uses the normal make command.
https://github.com/buserror/simavr/blob/master/Makefile.common#L147
Main binaries get build but the packages inside https://github.com/buserror/simavr/tree/master/tests fail.
(basically all .c files that use AVR_MCU_VCD_SYMBOL)

close2, 2015-09-25:
For those looking for a (temporary) workaround:
copy avr_mcu_section.h and change `char string[];` to `char string[128];` and `char name[];` to `char name[128];`
gcc will then know the size of string and name and no longer complain.

Obviously 128 is then the max size for your string/ name.

douzepouze, 2015-10-20:
I can confirm the issue on current arch linux and avr-gcc (GCC) 5.2.0.

Cynerd, 2016-03-09:
Will there be any terminal solution? This bug is really annoying. It is not exactly problem for me, but not be able build whole project is weird. I use avr-gcc (GCC) 5.3.0.

davidkirwan, 2016-05-21:
Using close2's workaround thanks!

sjlongland, 2016-06-12:
That worked around the problem for me under Gentoo (host GCC 4.9.3; `avr-gcc` 5.3.0).  In case people are wondering where this `avr_mcu_section.h` is and what to change:

```
diff --git a/simavr/sim/avr/avr_mcu_section.h b/simavr/sim/avr/avr_mcu_section.h
index 597858a..e849b30 100644
--- a/simavr/sim/avr/avr_mcu_section.h
+++ b/simavr/sim/avr/avr_mcu_section.h
@@ -82,7 +82,7 @@ struct avr_mmcu_long_t {
 struct avr_mmcu_string_t {
        uint8_t tag;
        uint8_t len;
-       char string[];
+       char string[128];
 } __attribute__((__packed__));
 
 struct avr_mmcu_addr_t {
@@ -96,7 +96,7 @@ struct avr_mmcu_vcd_trace_t {
        uint8_t len;
        uint8_t mask;
        void * what;
-       char name[];
+       char name[128];
 } __attribute__((__packed__));
 
 #define AVR_MCU_STRING(_tag, _str) \
```

spoorcc, 2016-12-03:
See [this stackoverflow answer](http://stackoverflow.com/a/27852112/1149326) which references the relevant part of C-standard why the construction is not allowed.

buserror, 2016-12-06:
That patch doesn't work anyway, well it works if you only have a single trace/string, as the parser for that section will fail to find further records in the section.

That construct worked fine for years, and only the new trend of making gcc more and more annoying to use seems to drive the fact it's now an error. The const char  string[] = "hello" worked for decades before this was deemed illegal. The string passed is not a variable, it's a known literal and it's not an affectation it's an initialization where the compiler knows /everything/ about the data being initialized.

So it's a compiler regression, it's not a bug in the code, regardless of stackunderflow's bit which doesn't apply in our case anyway.

hovercraft-github, 2017-01-10:
As a last resort for somebody, who is seeking workaround just to build simavr
and insists on using recent avr-gcc:
simply use
make build-simavr && make build-examples
instead of default target

buserror, 2017-02-16:
I've merged a fix for the compile, altho I'm not happy about it as it's a compiler regression... However, can someone confirms it works on the master branch?

dougszumski, 2017-02-26:
Thanks, with 29f4fdce i can build fine on master (gcc (GCC) 6.3.1 20161221 (Red Hat 6.3.1-1)).

buserror, 2017-02-26:
Thanks for reporting! I've also moaned to the GCC guys, whatever good that might do... :-)

closed 2017-02-26 by buserror