buserror/simavr #132: _MMCU_ from avr_mcu_section.h breaks curly braces initialization

easybe, 2015-09-16:
For simulating my code I used following lines and found some variables to be wrongly initialized.

    #include "avr_mcu_section.h"
    AVR_MCU(F_CPU, "attiny85");

Debugging the problem showed that `_MMCU_` attribute from `avr_mcu_section.h` was the cause of the problem.

    #include "avr_mcu_section.h"

    const uint8_t _mmcu[2] _MMCU_ = { 0, 0 };
    
    int main()
    {
        int arr[] = {1, 2, 3};
        return 0;
    }

The above test code will result in:

    arr = {2, 3, -1}

It seems to me that `__data_start` is off by two bytes.

I don't understand the simavr code too well, so I'm not able to provide a patch just now. Any hints are welcome though.

buserror, 2015-09-16:
Well it's not simavr code that is the problem here, it's the GNU binutils. The __MMCU__ tag tells the linker to put these two bytes into a separate ELF section, and it's supposed to make zero difference to the rest of the linked code. Basically _MMCU_ is just a directive, no special magic.

I haven't had access to 5.2 yet, but we've had reports that it doesn't like to compile the avr_mci_section.h in the first place.. not sure it's been tested that much it seems :-)

easybe, 2015-09-16:
OK, thanx for the quick feedback. In that case I will just not use avr_mcu_section.h...

buserror, 2015-09-16:
Note that that ELF section system has been in place for >5 years now, so if it didn't work, we'd know about it. They must have introduce some sort of bug...
For now you can for the MCU and frequency via command line parameters.. not idea but...

easybe, 2015-09-16:
So, I tried the same thing on my Mac with CrossPack 20131216 (avr-gcc 4.8.1, avr-binutils 2.23.2, avr-libc 1.8.0) and it wouldn't link at all:

    ld: section .mmcu loaded at [00000082,00000094] overlaps section .data loaded at [00000082,00000087]

The thing is, I hardly use any linker options, only `-Map` and I see that in the examples many are used. With `--gc-sections` it will link, but as far I can see still not work correctly.

If I test the problem with the example code on my Linux box everything works fine. So, I guess we are talking about a user error here and can close this (non-)issue.

The question is, am I missing some documentation? Or what must we add that this doesn't happens again?

buserror, 2015-09-16:
If you don't use gc-sections, the linker will probably assume that all sections are 'data' and will concatenate them.. (including the mmcu)
The simavr link options are made to be 'optimal' for AVR, ie make sure to remove most of the dead code, otherwise the 'general' linker will not do it (well, not enough when you try to save space)

Try to duplicate the link command, I /suspect/ you hit a linker problem here...

The /easiest/ way to use that link command is to copy/paste the Makefile.common rules, and rename your target as '.axf' -- I came up with that 'axf' BTW, it (probably) means 'AVR eXecutable Format' but is in fact plain EFF :-)

closed 2016-11-07 by buserror