buserror/simavr #150: Section offsets applied twice when debugging with gdb

stevedrake, 2015-12-18:
I am seeing an error when using gdb to debug a simple test program that has a global variable (initialised or uninitialised).  I am unable to inspect variables, including predefined ones such as PORTB, instead getting an error message such as:

    Cannot access memory at address 0x1000025

I'm using simavr master along with avr-gcc 4.9.1 and avr-gdb 7.6.1 from MacPorts.  Here is a test case that I can use to reproduce the problem (compiled with avr-gcc -O0 -g -mmcu=atmega48):
```c
#include <avr/io.h>
#include <util/delay.h>

uint8_t x = 0;

int main(void) {
    DDRB = 1<<DDB7;

    while (1) {
        PORTB ^= 1<<PB7;
        _delay_ms(x++);
    }
    return 1;
}
```

After starting simavr and loading the file in gdb, `info files` shows the following section info:
```
	0x00800100 - 0x00800100 is .data
	0x00000000 - 0x0000040a is .text
	0x00800100 - 0x00800101 is .bss
```
But after connecting to simavr with `target remote :1234`, the section info changes to:
```
	0x00800100 - 0x00800100 is .data
	0x00000000 - 0x0000040a is .text
	0x01000100 - 0x01000101 is .bss
```
At this point I'm not able to inspect global variables directly, but you can still see them with a bit of persistence:
```
(gdb) print PORTB
Cannot access memory at address 0x1000025
(gdb) print &PORTB
$1 = (uint8_t *) 0x800025 ""
(gdb) print *&PORTB
$3 = 0 '\000'
```

An adequate workaround for this problem is to (re)load the file into gdb after connecting to simavr.  The existing information is wiped away and reloaded directly from the file, and I haven't seen any further problems with that approach.

Another workaround I have tested successfully is to apply this patch:
```patch
diff --git a/simavr/sim/sim_gdb.c b/simavr/sim/sim_gdb.c
index e7d95ef..04bee1c 100644
--- a/simavr/sim/sim_gdb.c
+++ b/simavr/sim/sim_gdb.c
@@ -301,9 +301,9 @@ gdb_handle_command(
                                 */
                                gdb_send_reply(g, "1");
                                break;
-                       } else if (strncmp(cmd, "Offsets", 7) == 0) {
-                               gdb_send_reply(g, "Text=0;Data=800000;Bss=800000");
-                               break;
+                       // } else if (strncmp(cmd, "Offsets", 7) == 0) {
+                       //      gdb_send_reply(g, "Text=0;Data=800000;Bss=800000");
+                       //      break;
                        } else if (strncmp(cmd, "Xfer:memory-map:read", 20) == 0) {
                                snprintf(rep, sizeof(rep),
                                                "l<memory-map>\n"
```
It looks to me like specifying the offsets over the wire to gdb results in them being applied again to sections that already have the correct offsets.  Disabling this part of the response fixes the problem that I am seeing.  Not that I'm necessarily suggesting this as a fix, since presumably that code serves a purpose - but perhaps this will be helpful to someone who knows what they're looking at :)

buserror, 2015-12-22:
Well you might have picked up something behaving differently on more recent gdb/gcc -- that code was definitely needed before, but perhaps the offsets are now in a debug section in the elf and picked by gdb..
We'd probably need to figure out if it's the case...

Thanks for the detailed report too!

stevedrake, 2015-12-23:
Ah right, sounds reasonable...   Here's some more info that might help in testing your hypothesis (basically just a bunch of stuff that it occured to me to try out where 0x800000 pops up - and could be dependent on the toolchain versions?)

Toolchain, via MacPorts:
```
  avr-binutils @2.25_0 (active)
  avr-gcc @4.9.1_0+lto (active)
  avr-gdb @7.6.1_0 (active)
  avr-libc @1.8.1_0 (active)
  libelf @0.8.13_2 (active)
```

Output of avr-gcc -v (note `-Tdata 0x800100` linker option):
```
$ avr-gcc -v -O0 -g -mmcu=atmega48 -DF_CPU=1000000 test.o -o test
Using built-in specs.
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/opt/local/libexec/gcc/avr/4.9.1/lto-wrapper
Target: avr
Configured with: /opt/local/var/macports/build/_opt_mports_dports_cross_avr-gcc/avr-gcc/work/gcc-4.9.1/configure --prefix=/opt/local --target=avr --infodir=/opt/local/share/info --mandir=/opt/local/share/man --datarootdir=/opt/local/share/avr-gcc --with-system-zlib --with-gmp=/opt/local --with-mpfr=/opt/local --with-mpc=/opt/local --enable-stage1-checking --enable-multilib --enable-lto --enable-languages=c,c++
Thread model: single
gcc version 4.9.1 (GCC) 
COMPILER_PATH=/opt/local/libexec/gcc/avr/4.9.1/:/opt/local/libexec/gcc/avr/4.9.1/:/opt/local/libexec/gcc/avr/:/opt/local/lib/gcc/avr/4.9.1/:/opt/local/lib/gcc/avr/:/opt/local/lib/gcc/avr/4.9.1/../../../../avr/bin/
LIBRARY_PATH=/opt/local/lib/gcc/avr/4.9.1/avr4/:/opt/local/lib/gcc/avr/4.9.1/../../../../avr/lib/avr4/:/opt/local/lib/gcc/avr/4.9.1/:/opt/local/lib/gcc/avr/4.9.1/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-v' '-O0' '-g' '-mmcu=atmega48' '-D' 'F_CPU=1000000' '-o' 'test'
 /opt/local/libexec/gcc/avr/4.9.1/collect2 -plugin /opt/local/libexec/gcc/avr/4.9.1/liblto_plugin.so -plugin-opt=/opt/local/libexec/gcc/avr/4.9.1/lto-wrapper -plugin-opt=-fresolution=/var/folders/3r/8cts_5_95913tqj386xt0vhc0000gp/T//ccW6EEKv.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lm -plugin-opt=-pass-through=-lc -m avr4 -Tdata 0x800100 -o test /opt/local/lib/gcc/avr/4.9.1/../../../../avr/lib/avr4/crtm48.o -L/opt/local/lib/gcc/avr/4.9.1/avr4 -L/opt/local/lib/gcc/avr/4.9.1/../../../../avr/lib/avr4 -L/opt/local/lib/gcc/avr/4.9.1 -L/opt/local/lib/gcc/avr/4.9.1/../../../../avr/lib test.o --start-group -lgcc -lm -lc --end-group
```

Output of avr-objdump -h for the executable:
```
test:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000000  00800100  0000040e  000004a2  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  1 .text         0000040e  00000000  00000000  00000094  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .bss          00000001  00800100  00800100  000004a2  2**0
                  ALLOC
  3 .stab         000006c0  00000000  00000000  000004a4  2**2
                  CONTENTS, READONLY, DEBUGGING
  4 .stabstr      00000cdf  00000000  00000000  00000b64  2**0
                  CONTENTS, READONLY, DEBUGGING
  5 .comment      00000011  00000000  00000000  00001843  2**0
                  CONTENTS, READONLY
  6 .debug_info   000005f4  00000000  00000000  00001854  2**0
                  CONTENTS, READONLY, DEBUGGING
  7 .debug_abbrev 000005a2  00000000  00000000  00001e48  2**0
                  CONTENTS, READONLY, DEBUGGING
  8 .debug_line   0000001d  00000000  00000000  000023ea  2**0
                  CONTENTS, READONLY, DEBUGGING
  9 .debug_str    00000209  00000000  00000000  00002407  2**0
                  CONTENTS, READONLY, DEBUGGING
```

I hope that helps - please let me know if you'd like me to test out anything else.

hovercraft-github, 2016-10-21:
I have included a stevedrake's patch in my PR#180, because it completely satisfy me. But maybe it would be better to make it dependent on the GDB version? Or make it an option, which user can turn on by will? What do you think?

PR #180 (2016-10-24): sim_gdb: fix for restore callbacks issue
Fixes #179, fixes #150 .
These changes are of high importance to me, because I'm writing GUI wrapper for simavr: ([see discussion there](https://groups.google.com/forum/#!topic/simavr/94w3SzSsNgc), where user can start and stop debugging repeatably.
closed 2016-10-24 by buserror