buserror/simavr #157: Hooking up UART to stdin/stdout?

tengyifei, 2016-02-03:
Hi, I found this simulator very useful, but I couldn't figure out a way to send/receive data via the UART from the host computer to the microcontroller. Is there a way to do it?

augusteiner, 2016-02-03:
I've also been trying to do the same without much success anyway :'(

At last using picocom on linux I could see data flowing to the mcu (with the board_simduino example)

ricardojlrufino, 2016-11-06:
Use: board_simduino example

> export SIMAVR_UART_XTERM=1
> ./obj-x86_64-linux-gnu/simduino.elf File.cpp.hex 

this will create virtual/serial port: **/tmp/simavr-uart0**

Harvie, 2019-01-10:
@ricardojlrufino is there reason why "simavr" command can't do this? I am happy with following command:

`simavr -f 16000000 -m atmega328p grbl.hex `

It starts to print out UART output to console, but it cannot accept any input, nor it does create virtual serial port. I would love to have this in simavr directly... Both STDIN/OUT mode and virtual port, depending on commandline arguments.

MarcinWieczorek, 2019-02-07:
I'd love to see that feature too.