buserror/simavr #159: installation instructions for ubuntu incomplete

LieBtrau, 2016-02-09:
Following packages also need to be installed: 
sudo apt-get install freeglut3-dev libelf-dev

hydrocat, 2016-03-23:
did any of you made the display work ?

LieBtrau, 2016-03-23:
What do you mean?

hydrocat, 2016-03-28:
I mean that I couldn't get the emulated ledramp or the HD44780 LCD Board Demo to run or to show anything on the screen. I could only emulate the code and use gdb-avr on it (also, I couldn't get gdb to print the PORTD  or any io pin at all..)

LieBtrau, 2016-03-29:
This was not what my remark was about.  
I'm using Linux. I couldn't compile simavr from source.  The build failed because some packages were missing.  I had to install the above mentioned packages first.
I only ask you to update the installation instructions or the the build script.

BTW: I used simavr for simulating an attiny.  My application didn't involve any LCD.

closed 2017-07-29 by LieBtrau