buserror/simavr #162: Fast PWM mode not working

donothingloop, 2016-04-13:
Hi,

It seems that there is an issue with Fast PWM mode at least on the ATMEGA1280 core.

```c
#include <avr/io.h>

void main(void) {

        TCCR4A |= (1<<COM4A1) | (1<<WGM40) | (1<<WGM41);
        TCCR4B |= (1<<CS42);
        OCR4A = 120;
        DDRH = 255;
}
```
This works on the real hardware work but outputs the following error on simavr:
```
Load HEX flash 00000000, 306
firmware f-1460552902.hex f=16000000 mmcu=atmega1280
TIMER: avr_timer_write-4 clock turned off
TIMER: avr_timer_reconfigure-4 unsupported timer mode wgm=3 (0)
TIMER: wgm_op_mode_kind: 0
TIMER: avr_timer_write_ocr-4 mode 0 UNSUPPORTED
TIMER: avr_timer_reconfigure-4 unsupported timer mode wgm=3 (0)
TIMER: wgm_op_mode_kind: 0
```
As far as I understand avr_timer_wgm_fast_pwm is defined as **4** in *simavr/avr_timer.h* but according to the datasheet [www.atmel.com/images/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf] (page 184) it should be **5**.

Thank you very much

donothingloop, 2016-04-13:
It seems that I misread the datasheet. Nevertheless the behaviour of the real hardware is different than the simulator.

markasoftware, 2020-05-18:
simavr does not simulate the pin changes due to fast PWM; i.e, no bitbanging.