buserror/simavr #168: Tried the test and nothing happened

liborw, 2016-08-17:
Hi, I have tried the test from the README:

```
$ ./simavr/run_avr tests/atmega88_example.axf
AVR_MMCU_TAG_VCD_TRACE 00c6:00 - UDR0
AVR_MMCU_TAG_VCD_TRACE 00c0:20 - UDRE0
Loaded 1722 .text at address 0x0
Loaded 114 .data
Loaded 4 .eeprom
```

the produced `gtkwave_trace.vcd` is empty.

Pavel-Pavlov-ROI, 2016-08-18:
> the produced gtkwave_trace.vcd is empty.

Try adding signals to the timeline
I was also confused because it does not print further information eg

> Starting atmega88 - flashend 1fff ramend 04ff e2end 01ff
atmega88 init
avr_eeprom_ioctl: AVR_IOCTL_EEPROM_SET Loaded 4 at offset 0
Creating VCD trace file 'gtkwave_trace.vcd'
Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
Read from eeprom 0xcafef00d -- should be 0xcafef00d..
simavr: sleeping with interrupts off, quitting gracefully

Is it a bug or..? I see in version 1.0 it was ok.

freez10, 2016-08-18:
> the produced gtkwave_trace.vcd is empty.

Try adding signals to the timeline
I was also confused because it does not print further information eg

> Starting atmega88 - flashend 1fff ramend 04ff e2end 01ff
atmega88 init
avr_eeprom_ioctl: AVR_IOCTL_EEPROM_SET Loaded 4 at offset 0
Creating VCD trace file 'gtkwave_trace.vcd'
Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
Read from eeprom 0xcafef00d -- should be 0xcafef00d..
simavr: sleeping with interrupts off, quitting gracefully

Is it a bug or..? I see in version 1.0 it was ok.

Lotharyx, 2016-10-11:
Me too.  No output past "loaded 4 .eeprom":
```
$ simavr/run_avr tests/atmega88_example.axf 
AVR_MMCU_TAG_VCD_TRACE 00c6:00 - UDR0
AVR_MMCU_TAG_VCD_TRACE 00c0:20 - UDRE0
Loaded 1706 .text at address 0x0
Loaded 114 .data
Loaded 4 .eeprom
```
but the gtkwave_trace.vcd file seems whole.  Adding a couple `-v` flags to the command gave additional output:
```
$ simavr/run_avr -v -v -v -v tests/atmega88_example.axf 
AVR_MMCU_TAG_VCD_TRACE 00c6:00 - UDR0
AVR_MMCU_TAG_VCD_TRACE 00c0:20 - UDRE0
Loaded 1706 .text at address 0x0
Loaded 114 .data
Loaded 4 .eeprom
_avr_vcd_notify trace buffer resized to 51
Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
Read from eeprom 0xcafef00d -- should be 0xcafef00d..
simavr: sleeping with interrupts off, quitting gracefully
```
...but still not really the same thing as what's depicted in the README.

closed 2018-01-10 by buserror

AE1020, 2021-04-14:
I built on Linux (with plain `make`), and also do not get a match for what is in the output from the README.  Instead I get:

    $ simavr/run_avr tests/atmega88_example.axf 
    Loaded 1760 .text at address 0x0
    Loaded 114 .data
    Loaded 4 .eeprom
    Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
    Read from eeprom 0xcafef00d -- should be 0xcafef00d..

Adding -v doesn't change it to add more output.  This is in contrast to what the README says; which has additional lines:

    $ ./simavr/run_avr tests/atmega88_example.axf
    AVR_MMCU_TAG_VCD_TRACE 00c6:00 - UDR0
    AVR_MMCU_TAG_VCD_TRACE 00c0:20 - UDRE0
    Loaded 1780 .text
    Loaded 114 .data
    Loaded 4 .eeprom
    Starting atmega88 - flashend 1fff ramend 04ff e2end 01ff
    atmega88 init
    avr_eeprom_ioctl: AVR_IOCTL_EEPROM_SET Loaded 4 at offset 0
    Creating VCD trace file 'gtkwave_trace.vcd'
    Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
    Read from eeprom 0xcafef00d -- should be 0xcafef00d..
    simavr: sleeping with interrupts off, quitting gracefully

It matches for the lines that are present, but is missing several lines.

It produces a **gtkwave_trace.vcd**.  But viewing it with **gtkwave gtkwave_trace.vcd** does not look like the output in the sample.  :-/

> @buserror buserror closed this on Jan 10, 2018

Was this issue closed in error, or was it believed that a problem was fixed?