buserror/simavr #170: Make error about fatal error: libelf.h: No such file or directory

TestNemo, 2016-09-07:
Hi all,
I have one problem during my compiling the source code, the error is listed as bellow:

[nemo@localhost simavr]$ make
make -C simavr RELEASE=0
make[1]: Entering directory '/home/nemo/extra/simavr/simavr/simavr'
make obj config
make[2]: Entering directory '/home/nemo/extra/simavr/simavr/simavr'
make[2]: Nothing to be done for 'obj'.
make[2]: Nothing to be done for 'config'.
make[2]: Leaving directory '/home/nemo/extra/simavr/simavr/simavr'
make libsimavr run_avr
make[2]: Entering directory '/home/nemo/extra/simavr/simavr/simavr'
sim/sim_elf.c:32:20: fatal error: libelf.h: No such file or directory
compilation terminated.
../Makefile.common:187: recipe for target 'obj-x86_64-redhat-linux/sim_elf.o' failed
make[2]: *** [obj-x86_64-redhat-linux/sim_elf.o] Error 1
make[2]: Leaving directory '/home/nemo/extra/simavr/simavr/simavr'
Makefile:34: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/nemo/extra/simavr/simavr/simavr'
Makefile:15: recipe for target 'build-simavr' failed
make: *** [build-simavr] Error 2

mali, 2016-09-09:
Searching for "fedora 22 libelf.h" says that the package name is [elfutils-libelf-devel](http://rpm.pbone.net/index.php3/stat/4/idpl/34435393/dir/fedora_22/com/elfutils-libelf-devel-0.166-1.fc22.x86_64.rpm.html)

TestNemo, 2016-09-13:
Hi Navet, 
    You are right, it is fine after install  elfutils-libelf-devel on fedora 22.

closed 2016-09-13 by TestNemo

Grazfather, 2018-03-04:
`sudo apt install libelf-dev` on ubuntu

ZionChen01, 2023-08-06:
What is the name of this package in SUSE15?

edgar-bonet, 2023-08-06:
@ZionChen01: https://pkgs.org/search/?q=libelf  
Scroll down to “openSUSE”

alexander-naumov, 2024-05-10:
> What is the name of this package in SUSE15?
`sudo zypper in libelf-devel`

alexander-naumov, 2024-05-10:
> @ZionChen01: https://pkgs.org/search/?q=libelf Scroll down to “openSUSE”

well, openSUSE != SLES ;-)

newordersk, 2024-09-10:
Debian12  
sudo apt install libelf-dev

SJHust, 2025-08-01:
> Searching for "fedora 22 libelf.h" says that the package name is [elfutils-libelf-devel](http://rpm.pbone.net/index.php3/stat/4/idpl/34435393/dir/fedora_22/com/elfutils-libelf-devel-0.166-1.fc22.x86_64.rpm.html)

This works perfectly when you are using dnf or yum, instead of apt