buserror/simavr #182: Question: Is there a simple way for serial input?

aanno, 2016-10-29:
Question: Is there a simple way for serial input?

I fiddled around with your amazing simavr a bit and learnt the you could see the serial output of a simulation:

```
~/compile/simavr/simavr/run_avr -v -v -v -mcu atmega328 -freq 800000000 -t Release/qpn_pelican.elf
Loaded 5332 .text at address 0x0
Loaded 110 .data
TIMER: avr_timer_write-0 clock turned off
TIMER: avr_timer_write-0 clock turned off
TIMER: avr_timer_configure-0 TOP 390625.00Hz = 2048 cycles = 2usec
TIMER: avr_timer_configure-0 TOP 48828.12Hz = 16384 cycles = 20usec
TIMER: avr_timer_configure-1 TOP 1525.88Hz = 524288 cycles = 655usec
TIMER: avr_timer_configure-1 TOP 190.73Hz = 4194304 cycles = 5242usec
TIMER: avr_timer_reconfigure-1 unsupported timer mode wgm=1 (0)
TIMER: avr_timer_configure-2 TOP 48828.12Hz = 16384 cycles = 20usec
TIMER: avr_timer_reconfigure-2 unsupported timer mode wgm=1 (0)
ADC: Start AREF 0 AVCC 0
UART: 0 configured to 0010 = 2941176 bps (x1), 8 data 1 stop
UART: Roughly 3 usec per bytes
Start, QP-nano version: 562..
TIMER: avr_timer_configure-2 TOP 12500000.00Hz = 64 cycles = 0usec
TIMER: avr_timer_configure-2 TOP 781250.00Hz = 1024 cycles = 1usec
TIMER: avr_timer_configure-2 TOP 5008.01Hz = 159744 cycles = 199usec
Cars: RED..
Peds: DON'T WALK..
Cars: GREEN..
State->carsGreenNoPed..
State->carsGreenInt..
```

However I wonder if there is also a simple way for serial input to the simulation. It seems not to be connected to stdin...

buserror, 2016-11-01:
Well, the *easiest* way would be to run your firmware into the 'simduino' example, as it has a way of forking a 'real' terminal and redirect the serial port in/out there (via a PTY)...

hovercraft-github, 2016-11-03:
Also, you can try the [simutron](https://sourceforge.net/projects/simutron/) - a GUI wrapper for simavr, where you can just to draw your test circuit, drag the UART - TTY link component, and that's it.

ricardojlrufino, 2016-11-06:
@hovercraft-github I'm trying to use your simulator on Ubuntu x64 but I'm not getting it. It looks like the lib: libQt5Widgets.so.5 is missing
Already Installed: qt5-default cutecom gtkwave
Would you like to make the x64 version available?

ricardojlrufino, 2016-11-06:
@hovercraft-github I can compile, by running the build unix script, please update the HELP by putting this guide.

hovercraft-github, 2016-11-06:
@ricardojlrufino : Thank you for your interest. I will create wiki page about compiling soon. Could you please report in more details, do you  successfully compile it and what problems besides lack of help you met.
BTW you will be probably the first who built it on the 64 bit platform :)

hovercraft-github, 2016-11-06:
06.11.2016 23:42, Ricardo JL Rufino пишет:

>
> @hovercraft-github <https://github.com/hovercraft-github> I can
> compile, by running the build unix script, please update the HELP by
> putting this guide.
Also the trunk version contains more plugins, namely hd44780 and pcd8544
LCD and pushbutton which are not yet in the build script.

ricardojlrufino, 2016-11-07:
The problem I had was to apply the "PATCH" of ARVSIM as explained in the README. 
I could only compile through the script. And its working,

I think it's interesting to launch the x64 version, as I think most use it.

hovercraft-github, 2016-11-07:
This is because readme outdated - this patch is already applied in the
simavr master branch.
Also several hours ago some other changes made which can (maybe) help in
your case,
so I would advise you to get fresh simavr sources and recompile it. This
may make simavr less restrictive with respect to the target MCU address
range.

07.11.2016 23:03, Ricardo JL Rufino пишет: