buserror/simavr #183: ATMEGA328p - avr_sadly_crashed

ricardojlrufino, 2016-11-06:
Running...
> simavr -m atmega328p UsbConnection.ino.elf

I got this error:

> CORE: *** Invalid read address PC=1cc2 SP=08e4 O=918c Address 859e out of ram (08ff)
avr_sadly_crashed
avr_gdb_init listening on port 1234
^Csignal caught, simavr terminating

I compiled using Arduino IDE - Uno Board.
I am sending the folder that Arduino generates with the compiled files

[USBTest.tmp.zip](https://github.com/buserror/simavr/files/573898/USBTest.tmp.zip)

hovercraft-github, 2016-11-07:
Well, this usually just means there is a bug in you firmware. Under this condition simavr activates the GDB backend and waits debugger connection using TCP on the localhost:1234. You can setup your IDE (Eclipse or CodeBlocks ..) to start the 'remote' debug session using this address, and then trace the situation leading to the above error.

ricardojlrufino, 2016-11-07:
It was what I imagined at first, but the same code is running from Arduino ONE.
I think it might be some mistake to access the EEPROM, I'll investigate more calmly, trying to debug.

closed 2018-01-10 by buserror