buserror/simavr #186: ATMega1284p core crashes while ATMega328p is ok

wimalopaan, 2016-11-11:
The following programm produces the error

CORE: *** 00a8: Invalid Opcode SP=40ff O=9007 

when used with atmega1284p core:

`
#include <stdint.h>
#include <stdbool.h>

volatile uint8_t global = 3; //-
volatile bool b = false;
volatile bool c = false;

//[value
bool fooA(uint8_t x) {
    global = x; //-
    return b;
}
bool fooB(uint8_t x, uint8_t y) {
    global = x + y; //-
    return b;
}
//]
//[main
int main()
{
    uint8_t a = 42;
    uint8_t b = 43;

    c = fooA(a);
    c = b && fooB(a, b);

    while(true);
}
//]
`

buserror, 2016-11-14:
What toolchain/host os are you using?

wimalopaan, 2016-11-14:
avr-g++ 6.2.1
linux host (arch linux)

buserror, 2016-11-14:
Ok so it seems we had't specified the RAMPZ register in the core definition. I've just pushed a fix, it should make it all work now. That opcode is ELPM, Extended Load Program Memory.

wimalopaan, 2016-11-14:
Looks good!

closed 2016-11-14 by wimalopaan