buserror/simavr #189: UCSR0A is not refreshed when writing on uart_pty like in simduino example to emulate UART RX on Arduino side

bvernoux, 2016-12-06:
UCSR0A is not refreshed when writing on uart_pty like in simduino example to emulate UART RX on Arduino side.
With no patch when a normal program poll UART for data received, by checking UCSR0A/0xC0  until "USART Receive Complete" bit is set the it will do an infinite loop ... (even if some characters are received correctly in the fifo => UDR0/0xC6)

In order to fix that I have done an ugly patch in simavr/sim/avr_uart.c by replacing avr_uart_rxc_raise function by 
```
static avr_cycle_count_t avr_uart_rxc_raise(struct avr_t * avr, avr_cycle_count_t when, void * param)
{
	avr_uart_t * p = (avr_uart_t *)param;
	if (avr_regbit_get(avr, p->rxen)) {
		/* Force UCSR0A/0x00C0 to 0xe0 "data ready" => Bit 7 – RXC0: USART Receive Complete |  Bit 6 – TXC0: USART Transmit Complete | Bit 5 – UDRE0: USART Data Register Empty */
		avr_core_watch_write(avr, 0x00C0, 0xe0);
		avr_raise_interrupt(avr, &p->rxc);
	}
	return 0;
}
```
A better fix will be probably to do a avr_core_watch_read() of UCSR0A (0xC0) reg and then set the correct bits and then do an avr_core_watch_write()
I imagine UCSR0B/0xC1 is also probably not refreshed correctly

pakesson, 2016-12-11:
I have an alternative workaround at pakesson/simavr@8467783d44dc5b36a7b1fcb8c70403238f41ce1a. It's a slightly cleaner solution (no hardcoded register bits), but may have unintended side effects that I haven't considered.

buserror, 2016-12-11:
@pakesson if you do a bit more testing and are happy, I'd be happy to merge your patch... thanks for posting

pakesson, 2016-12-12:
@buserror I have tested three different ATmega328 programs as well as one for the ATmega128rfa1, and they appear to work fine (with UART input and output) after the patch.

I don't know all the inner workings of simavr well enough to say for certain that there are no regressions for other cores or use cases, but if you think the patch looks good then please feel free to merge it :)

buserror, 2016-12-13:
Hmm actually this breaks the tests with tests/*_uart* so there must be soemthing wrong there.

pakesson, 2016-12-13:
Strange, the tests are running fine here (with the 'run_tests' makefile target)

hovercraft-github, 2017-02-15:
This probably can be closed now, because #190 and #194 fixed it.

closed 2017-02-16 by buserror