buserror/simavr #205: build error: "initialization of flexible array member in a nested context"

bjornfor, 2017-02-28:
Hi,

I noticed simavr is broken in the Nix package manager. It looks like the code is using a construct that gcc doesn't accept anymore (from what I can tell):

```
$ nix-build -A simavr
...
AVR-CC atmega32_ssd1306.c
In file included from atmega88_uart_echo.c:22:0:
atmega88_uart_echo.c:34:23: error: initialization of flexible array member in a nested context
  { AVR_MCU_VCD_SYMBOL("UDR0"), .what = (void*)&UDR0, },
                       ^
```

A quick seach led me here: http://stackoverflow.com/questions/3875523/lookup-table-in-c

bjornfor, 2017-02-28:
CC @goodrone (maintainer of the package in Nixpkgs).

buserror, 2017-03-01:
This is already fixed in the master branch.

bjornfor, 2017-03-01:
Great! When will a release with that fix be ready?

buserror, 2017-03-01:
Not sure; do you need a tag to do that?

bjornfor, 2017-03-01:
Yes, a git tag is what I had in mind. I see you have releases/tags already at https://github.com/buserror/simavr/releases. I'm simply looking for a v1.5 or something, that includes the fix, which we can add to Nixpkgs.

buserror, 2017-03-01:
Ok, well unless someone has an objection, this version appear quite stable as it is, and I have a few 'bigger' bits to push later, so I'll tag...

closed 2017-03-01 by buserror

bjornfor, 2017-03-01:
Thank you! Nixpkgs now has v1.5 (working).