buserror/simavr #208: sigrok with simavr vcd

husnoo, 2017-03-13:
Hi,

On the off chance anyone has any ideas... I've got a VCD from simavr:

```
$timescale 1ns $end
$scope module logic $end
$var wire 8 ! PORTB $end
$upscope $end
$enddefinitions $end
$dumpvars
bxxxxxxxx !
$end
#2601250
b00000000 !
.
.
.
```

when I try to open it with pulseview, I get "Floating point exception" (the only error), and with sigrok-cli, I get 

```
sr: input/trace32_ad: Don't know how to handle this file with record size 32.
Floating point exception
```

If anyone knows enough about either the VCD format and how closely simavr adheres, how sigrok deals with VCDs, I'd be grateful for any pointers!

thanks

Nawal

buserror, 2017-03-13:
I already had a chat with the sigrock guys last week, they mentioned 'vectors' aren't supported when reading VCD's -- note that that file opens just fine in gtkwave (I used that primarily for testing).

Also, when *reading* a VCD file, simavr *should* be just file reading pulseview's format, as it's what I've used for testing...

husnoo, 2017-03-13:
Hi, thanks for the quick reply, and thanks even more for simavr, it's a great project!

Do you have any suggestions on what I would need to change in the file to make it readable by sigrok? What does this "vector" thing refer to in the data I shared above? I'd be interested in writing a simple function in python to transform the simavr vcd into something sigrok can read.

thanks

husnoo, 2017-03-13:
Hi, thanks for the quick reply, and thanks even more for simavr, it's a
great project!

Do you have any suggestions on what I would need to change in the file to
make it readable by sigrok? What does this "vector" thing refer to in the
data I shared above? I'd be interested in writing a simple function in
python to transform the simavr vcd into something sigrok can read.

thanks

Nawal.

buserror, 2017-03-13:
Instead of using the whole PORTB (that will include a 'vector' of 8 bits) try to use individual bits for each pins... that should be supported by pulseview...

husnoo, 2017-03-13:
oh cool, thanks, will give that a go!

husnoo, 2017-03-13:
That worked :-D
 thanks!

closed 2017-03-13 by husnoo