buserror/simavr #210: USI on attiny85

husnoo, 2017-03-24:
Hi,

Just checking: does simavr handle the USI overflow interrupt on ATtiny85? My code below never blinks...

thanks!

```
#include <avr/io.h>
#include <avr/interrupt.h>

#define F_CPU 8000000
#include "avr_mcu_section.h"
AVR_MCU(F_CPU, "attiny85");

const struct avr_mmcu_vcd_trace_t _mytrace[]  _MMCU_ = {
    { AVR_MCU_VCD_SYMBOL("PB2"), .mask = (1 << PB2), .what = (void*)&PORTB, },    
};

#define TIMER_TOP 100

volatile int x;

ISR(TIMER0_COMPA_vect) {
    x = x + 1;
    if (x==12) {
	USISR = 1<<USIOIF | (16 - 8);
	x = 0;
    }
    return;
}

ISR(USI_OVF_vect) {
    PORTB = PORTB ^ _BV(PB2);	  
    return;
}

    
int main (void) {
    x = 0;
    OCR0A = TIMER_TOP; // interrupt when output compare
    TIMSK = 1<<OCIE0A; // interrupt compare match A enable
    TCCR0A = 1<<WGM01;           // CTC mode
    TCCR0B = 1<<CS01;        // Set prescaler to clk or clk /8
    GTCCR |= 1 << PSR0;           // Reset prescaler
    TCNT0 = 0;                    // Count up from 0
    // Select three wire mode so USI output on PB1
    // Enable USI Counter OVF interrupt.
    // Timer0 Compare Match as USI clock source
    USICR = (0<<USIWM1)|(1<<USIWM0)|(1<<USIOIE)| (0<<USICS1)|(1<<USICS0)|(0<<USICLK);
    DDRB |= (1<<PB1); // Configure USI DO, PB1 as output
    DDRB |= (1<<PB2); // test blink
    sei();    
    while (1) {

    }
}
```

husnoo, 2017-03-24:
Compiled with

```
avr-gcc -o three4th.out  \      
        -Wall -gdwarf-2 -Os -std=gnu99 \
        -mmcu=attiny85 \        
        -fno-inline-small-functions \
        -ffunction-sections -fdata-sections \
        -Wl,--relax,--gc-sections \
        -Wl,--undefined=_mmcu,--section-start=.mmcu=0x910000 \
        three4th.c \            
        -I/nix/store/25w4x6ggjzaq8cjjsx60bx7yw07h447x-simavr-1.3/include/simavr/avr/
   
```

husnoo, 2017-03-24:
Attempting this on an actual chip shows the overflow interrupt does fire, so I suspect this means simavr doesn't implement it...

gherlein, 2018-09-30:
Is this still the case?

JarrettBillingsley, 2019-03-23:
As far as I can tell, there's no code in simavr to support the USI peripheral at all (!).

closed 2025-03-31 by gatk555