buserror/simavr #213: remote debug

wenhuizhang, 2017-04-21:
Hi, I tried remote debug, yet not sure how it works, any idea on which command I should  I use please?

```
wenhui@wenhui-XPS-8700:~/Downloads/simavr$ ./simavr/run_avr -m atmega88 -f 9600  /home/wenhui/Downloads/Arduino_Project/env_test/led.hex 
Loaded 1 section of ihex
Load HEX flash 00000000, 306
avr_gdb_init listening on port 1234
gdb_network_handler connection opened
gdb_network_handler connection closed

```

```
wenhui@wenhui-XPS-8700:~/Downloads/Arduino_Project/env_test$ gdb
GNU gdb (Ubuntu 7.7-0ubuntu3.1) 7.7
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote 127.0.0.1:1234
Remote debugging using 127.0.0.1:1234
Reply contains invalid hex digit 59

```

buserror, 2017-04-24:
Won't work with 'gdb' you need to use 'avr-gdb' -- As the message **This GDB was configured as "x86_64-linux-gnu"** suggest, it won't grock AVR.

closed 2017-04-24 by buserror

wenhuizhang, 2017-04-25:
Got it, thanks for your insights and suggestions , I will give it a try and let you know how it goes

wenhuizhang, 2017-04-25:
Hi， thanks so much , @buserror , I tried to connect it to avr-gdb, still got some issues, any suggestions?
```c
// test: led.c
#ifndef F_CPU
#define F_CPU 16000000UL // or whatever may be your frequency
#endif
 
#include <avr/io.h>
#include <util/delay.h>                // for _delay_ms()
 
int main(void)
{
    DDRC = 0x01;                       // initialize port C
    while(1)
    {
        // LED on
        PORTC = 0b00000001;            // PC0 = High = Vcc
        _delay_ms(500);                // wait 500 milliseconds
 
        //LED off
        PORTC = 0b00000000;            // PC0 = Low = 0v
        _delay_ms(500);                // wait 500 milliseconds
    }
}
```
```bash
avr-gcc -g -Os -mmcu=atmega2560 -c led.c
avr-gcc -g -mmcu=atmega2560 -o led.elf led.o
```

```bash
client-130-203-90-97:simavr WenhuiZhang$ sudo ./run_avr -m atmega2560 -f 9600  -g  /Users/WenhuiZhang/Downloads/Arduino_Project/env_test/led.hex  
Loaded 1 section of ihex
Load HEX flash 00000000, 306
avr_gdb_init listening on port 1234
gdb_network_handler connection opened
```

## I got : 
```bash
(gdb) bt 
#0  0x00000000 in ?? ()
(gdb) bt
#0  0x00000000 in ?? ()
(gdb) down 
Bottom (innermost) frame selected; you cannot go down.
(gdb) down 
Bottom (innermost) frame selected; you cannot go down.
(gdb) up
Initial frame selected; you cannot go up.
(gdb) up
Initial frame selected; you cannot go up.
(gdb) frame 
#0  0x00000000 in ?? ()
(gdb) return 
Make selected stack frame return now? (y or n) y
Cannot access memory at address 0x802201
(gdb) fram e
No symbol table is loaded.  Use the "file" command.
(gdb) frame
#0  0x00000000 in ?? ()
(gdb) backtrace 
#0  0x00000000 in ?? ()
```

buserror, 2017-04-26:
Whats your problem there? you tell it to stop with -g, so it doesn't run and wait at the reset vector at 0x0... and there is no stack frame anyway, so 'down' 'up' and 'backtrace' won't do a thing...

try 'cont' then hit control-c... but then again there's only main is being called, and likely without a return address either, so you won't be able to see a stack there...

buserror, 2017-05-04:
That's not really a simavr bug either ;-)