buserror/simavr #217: simavr does not produce VCD file, freezes

bendavis78, 2017-05-01:
I have the following C code:

```c
#define F_CPU (int)1000000L
#include <avr/io.h>
#include <util/delay.h>
#include <simavr/avr/avr_mcu_section.h>

AVR_MCU(F_CPU, "attiny85");
AVR_MCU_VCD_FILE("my_trace_file.vcd", 1000);

const struct avr_mmcu_vcd_trace_t _mytrace[] _MMCU_ = {
	{ AVR_MCU_VCD_SYMBOL("PB4"), .mask = (1 << 4), .what = (void*)&PORTB, },
};

int main(void)
{
    DDRB = 1<<4; // PB4 is output
    PORTB = 0<<4; // PB4 low to start

    int i;
    for (i=0; i<5; i++) {
        PORTB = 1<<4; // set PB4 high
        _delay_ms(250); // wait some time
        PORTB = 0<<4; // set PB4 low
        _delay_ms(250); // wait some time
    }
    return 0;
}
```
I compiled a .hex with:

    $ avr-gcc -g -Os -Wall -mcall-prologues -mmcu=attiny85 blinky.c -o blinky
    $ avr-objcopy -R .eeprom -O ihex blinky blinky.hex

Then I tried run_avr, but it just freezes and does nothing:

    $ run_avr -m attiny85 -f 1000000 blinky.hex                                        
    Loaded 1 section of ihex
    Load HEX flash 00000000, 281
    _

bendavis78, 2017-05-02:
Just realized this is due to the fact that I didn't have `sleep_cpu()` at the end of my program.

closed 2017-05-02 by bendavis78

buserror, 2017-05-03:
Ah, that was an easy one. You can also control-C, the VCD file should be flushed properly, if you used run_avr