buserror/simavr #219: Update Travis CI to avrlibc 2.0

rkel, 2017-05-05:
I am currently working on Msys2 (windows) support for simavr. I have it up and running locally and I have did first [pull request to your repository](https://github.com/buserror/simavr/pull/218).
And I have a problem with your CI agent configuration. This is not an error in my code. It is avrlibc 1.8.0 bug. The problem is you are using avrlibc 1.8.0 that is 6 years old! Please update to avrlibc 2.0 (current - 1 year old) or at least to 1.8.1 (3 years old) where an error you can see in my pull request it fixed.

I have admit I do not know the Travis CI infrastructure - is is possible to update?

buserror, 2017-05-08:
I actually have no idea how to do that; I know that old ubugtu have old libraries like that, and people still use the damn things for years and years afterward.. will look into it...

yurayko, 2017-05-09:
About update Travis CI infrastructure (spoiler: no way)
https://github.com/travis-ci/travis-ci/issues/5821
and if its happened - maximum version is 1:1.8.0+Atmel3.5.0-1
(http://packages.ubuntu.com/xenial/avr-libc)

rkel, 2017-05-09:
Sorry - I have really little time for that in a middle of the week - I might try the whole Travis setting during the weekend or at some evening. Just searching very quickly I have found that PPA is allowed: [link](https://docs.travis-ci.com/user/installing-dependencies/#Installing-Packages-from-a-custom-APT-repository). Also deb files are possible to be used: [link](https://docs.travis-ci.com/user/installing-dependencies/#Installing-Packages-without-an-APT-Repository).
Would it help?

SGOrava, 2017-05-09:
You can use docker with travis

buserror, 2017-05-10:
Well docker or whatever won't help much, as it seems 1.8.0xxx is the latest even in debian sid. I think I might consider pulling the headers from avr-libc and merging them with simavr, that way, we'll control what 'cores' are available on every platform...

SGOrava, 2017-05-10:
In Arch is avrlibc 2.0

rkel, 2017-05-10:
Still see the last link I have provided. [Installing Projects from Source](https://docs.travis-ci.com/user/installing-dependencies/#Installing-Projects-from-Source) header. You can provide any script here that would even download, unpack and copy to the right folders the required files. It makes it possible to download the official [Atmel toolchain](http://www.atmel.com/tools/atmelavrtoolchainforlinux.aspx). I know it is pretty old, but at least it uses avrlibc 2.0.0 and many users, for Windows are downloading it by default with ATMEL studio.

I know that Windows is not the most favourite choose as a target platform for Open Source software, but it is still pretty popular and my current idea is to provide the software also for Windows users.

buserror, 2017-05-10:
What I'm planning is to add the AVR lib headers straigth to our own repository, no extra download needed. Given that /technically/ we don't need avr-gcc for simavr (just the header) I don't even need the library/code part...
Of course avr-gcc is needed to build the tests etc, but strictly speaking, it's not needed for the cores of simavr, just the headers. That would also simplify the makefile for many other platforms...

rkel, 2017-05-11:
I like this idea. Simple and effective.

buserror, 2017-08-03:
Well this has now been implemented, we're sorted!

closed 2017-08-03 by buserror