buserror/simavr #225: Apply CPPFLAGS to avr-gcc in Makefile.common?

tobz, 2017-05-16:
Hey there!

I'm trying to use your awesome project (super awesome, in fact) in my own project, and I'm having some trouble adapting all the Makefile-fu.

What I'd like to do is structure my project "normally", with code sort of at the top and dependencies in subfolders, in this case, simavr.. which will be checked out by some simple dependency setup scripting.

Currently, however, `Makefile.common` in simavr is setup with include paths for the avr-gcc call that assume you'll be calling it from a board folder in the examples.

Given that the Makefile for boards seem to at least know where they are, and are overriding IPATH/VPATH already, it'd be sweet if the common Makefile could do this for us, too!

I already tested shoving it in there, and it seems to work as long as I follow all the other naming conventions... but the intent is to check out simave when needed, and not maintain a fork of it.  I also suspect that others possibly care about this? for anyone running custom code around their simulation (aka ledramp style).

Unless you have/know of a way for me to emulate inputs (in this case, I'm wiring up a keypress to an external interrupt pin) so that I can ditch the custom code and just run it standalone with `run_avr`.... :D

tobz, 2017-05-16:
I suppose, too, that a third option would be trying to figure out if the common Makefile can be abstracted, like the core bits, even further... or have an example Makefile for my situation?  An all-in-one sort of thing where it's not dependent on the directory relationship to the simavr code.

I'm willing to do this on my own, as a PR even, but I'm hoping you have some sage advice on what you think the direction should be given the project's intent. :)

buserror, 2017-05-16:
Hi @tobz, it *should* work really, as I use it in multiple projects myself. One example is https://github.com/buserror/simreprap/blob/master/Makefile for example... tell me if this helps

closed 2018-01-10 by buserror