buserror/simavr #239: Compiling Error with -Wimplicit-fallthrough (default since avr-gcc 7x)

thecky, 2017-08-06:
I get the following error, when I try to compile the current master:

`sim/sim_core.c:978:6: Fehler: diese Anweisung könnte durchfallen [-Werror=implicit-fallthrough=]
      avr_interrupt_reti(avr);
      ^~~~~~~~~~~~~~~~~~~~~~~
sim/sim_core.c:979:5: Anmerkung: hier
     case 0x9508: { // RET -- Return -- 1001 0101 0000 1000
     ^~~~
cc1: Alle Warnungen werden als Fehler behandelt

[this statement may fall through]`

There's a missing "break;" inside the case statement or if this "fallthrough" is desired you should add a "-Wno-implicit-fallthrough" to the CFLAGS in the Makefile.common file.

buserror, 2017-08-07:
Likely will break with other previous gcc tho. I hate when compilers add 'new features' with these stupid warnings. It's /meant/ to be like that. Try to see if "if (0) break;" removes the warning perhaps?

thecky, 2017-08-07:
The "if (0) break" doesn't solve the error-warning.

But we could add a "__attribute__ ((fallthrough));" to the case statement (I can do a PR for all occurrences), that should solve/avoid the errors - and it remains readable.

ConnorRigby, 2017-08-07:
i was able to add `attribute ((fallthrough));` to a couple places that fixed this compiler error. I could PR if you are interested @buserror. Unfortunately i think it may break other/older compilers.

buserror, 2017-08-08:
Yeah well I removed -Werror, we just can't win with clang, gcc on different platforms disagreeing on stuff and breaking the build.

closed 2017-08-08 by buserror