buserror/simavr #242: ../examples/parts/obj-x86_64-apple-darwin16.6.0/libsimavrparts.a: No such file or directory

lynndylanhurley, 2017-08-09:
I ran the following commands (copied from the `README`):

~~~
brew tap osx-cross/avr
brew install --HEAD simavr
~~~

The install failed on [this error](https://gist.github.com/lynndylanhurley/9a71ee68232088b983e414a7c6106834).

Here are the last few lines for anyone else searching for this:

~~~
install: ../examples/parts/obj-x86_64-apple-darwin16.6.0/libsimavrparts.a: No such file or directory
make[1]: *** [install] Error 71
make: *** [install] Error 2
~~~

jacquesg, 2017-08-10:
This has been fixed by osx-cross/homebrew-avr#57

lynndylanhurley, 2017-08-10:
Just tried to reinstall, now I'm seeing this error:

~~~
==> Installing simavr from osx-cross/avr
==> Cloning https://github.com/buserror/simavr.git
Updating /Users/lynnhurley/Library/Caches/Homebrew/simavr--git
==> Checking out branch master
==> make all HOMEBREW_PREFIX=/usr/local RELEASE=1
Last 15 lines from /Users/lynnhurley/Library/Logs/Homebrew/simavr/01.make:
CC ssd1306demo.c
error: unable to open output file 'obj-x86_64-apple-darwin16.7.0/ssd1306demo.o': 'No such file or directory'
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin16.7.0/ssd1306demo.o] Error 1
make[2]: *** Waiting for unfinished jobs....
   1594	      0	   1024	   2618	    a3a	atmega32_ssd1306.axf
AVR-CC atmega168_timer_64led.c
CC timer_64led.c
error: unable to open output file 'obj-x86_64-apple-darwin16.7.0/timer_64led.o': 'No such file or directory'
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin16.7.0/timer_64led.o] Error 1
make[2]: *** Waiting for unfinished jobs....
   1900	     20	     53	   1973	    7b5	atmega168_timer_64led.axf
make[1]: *** [all] Error 2
make: *** [build-examples] Error 2
~~~

jacquesg, 2017-08-10:
Correct, see #244

vrt1shjwlkr, 2017-12-23:
Hi I am still facing the error while using make install RELEASE=1 as mentioned in the README. I am using Ubuntu 16.04. Output snippets are below:

Makefile:33: warning: overriding recipe for target 'install'
Makefile:27: warning: ignoring old recipe for target 'install'
make -C simavr install
make[1]: Entering directory '/home/vshejwalkar/Downloads/simavr/simavr'
make obj config
make[2]: Entering directory '/home/vshejwalkar/Downloads/simavr/simavr'
make[2]: Nothing to be done for 'obj'.
make[2]: Nothing to be done for 'config'.
make[2]: Leaving directory '/home/vshejwalkar/Downloads/simavr/simavr'
make libsimavr run_avr
make[2]: Entering directory '/home/vshejwalkar/Downloads/simavr/simavr'
make[2]: Nothing to be done for 'libsimavr'.
LD obj-x86_64-linux-gnu/run_avr.elf
ln -sf obj-x86_64-linux-gnu/run_avr.elf run_avr
make[2]: Leaving directory '/home/vshejwalkar/Downloads/simavr/simavr'
mkdir -p /usr/local/include/simavr/avr
mkdir -p /usr/local/include/simavr/parts
install -m644 sim/*.h /usr/local/include/simavr/
install -m644 sim_core_*.h /usr/local/include/simavr/
install -m644 sim/avr/*.h /usr/local/include/simavr/avr/
install -m644 ../examples/parts/*.h /usr/local/include/simavr/parts/
mkdir -p /usr/local/lib
install obj-x86_64-linux-gnu/libsimavr.a /usr/local/lib/
install ../examples/parts/obj-x86_64-linux-gnu/libsimavrparts.a /usr/local/lib/
i**nstall: cannot stat '../examples/parts/obj-x86_64-linux-gnu/libsimavrparts.a': No such file or directory**
Makefile:81: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/home/vshejwalkar/Downloads/simavr/simavr'
Makefile:33: recipe for target 'install' failed
make: *** [install] Error 2
 
Any help is appreciated. Thanks

gatk555, 2025-04-08:
Fix was merged,

closed 2025-04-08 by gatk555