buserror/simavr #263: Mac compile error due to: undeclared identifier 'CLOCK_MONOTONIC_RAW'

andi4s, 2017-11-02:
Hello,
first of all thx for your great work!
I tried to install simavr on my Macbook (Mac OS X El Capitan 10.11.6, Darwin Kernel Version 15.6.0)
using homebrew (brew install --HEAD simavr) but unfortunately I got this compile error:

------------------------------------------------------------------

==> make all HOMEBREW_PREFIX=/usr/local RELEASE=1
Makefile:33: warning: overriding commands for target `install'
Makefile:27: warning: ignoring old commands for target `install'
/Library/Developer/CommandLineTools/usr/bin/make -C simavr RELEASE=1
fatal: No names found, cannot describe anything.
/Library/Developer/CommandLineTools/usr/bin/make obj config
fatal: No names found, cannot describe anything.
make[2]: Nothing to be done for `obj'.
make[2]: Nothing to be done for `config'.
/Library/Developer/CommandLineTools/usr/bin/make libsimavr run_avr
fatal: No names found, cannot describe anything.
CC sim/sim_cmds.c
CC sim/sim_avr.c
CC sim/sim_core.c
CC sim/sim_cycle_timers.c
sim/sim_avr.c:154:16: error: use of undeclared identifier 'CLOCK_MONOTONIC_RAW'
        clock_gettime(CLOCK_MONOTONIC_RAW, &tp);
                      ^
sim/sim_avr.c:334:16: error: use of undeclared identifier 'CLOCK_MONOTONIC_RAW'
        clock_gettime(CLOCK_MONOTONIC_RAW, &tp);
                      ^
2 errors generated.
make[2]: *** [obj-x86_64-apple-darwin15.6.0/sim_avr.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [all] Error 2
make: *** [build-simavr] Error 2

------------------------------------------------------------------

Maybe somebody else also had (already) this issue (and can help)?

thx,
andi

buserror, 2017-11-05:
Hmm yeah it's that lastest commit we've got. Need to add a workaround for when that isn't defined; and use getlocaltime as a fallback

dxxb, 2017-12-04:
@buserror, or maybe just revert that commit and leave it to the user to override avr_callback_sleep_raw() if needed. A short _howto_ for overriding it could be added to the README for instance.

flocsy, 2017-12-10:
Is there an ETA when will this be fixed?

flocsy, 2017-12-11:
I created a pull request: https://github.com/buserror/simavr/pull/266

buserror, 2018-01-10:
I've pushed a 'proper' fix for that, please give feedback so i can close these issues

flocsy, 2018-01-11:
ok it compiles now

closed 2018-01-15 by buserror