buserror/simavr #269: The PC don't wraps around to zero in the end of flash

lcgamboa, 2018-01-02:
The PC don't wraps around to zero when the flash memory end. The problem is in the line: 

https://github.com/buserror/simavr/blob/6660aba00a898ff4db6c689e35e7fd03b4f9d47d/simavr/sim/sim_core.c#L645

To fix it I changed the code to:

``
avr_flashaddr_t	new_pc = (avr->pc + 2 )% (avr->flashend+1); // future "default" pc
``

Please update it in the code of master branch.

Best regards.

buserror, 2018-01-08:
Actually I don't like that patch. It introduces a modulo (divide) per instruction, for a case that is extraordinarily rare of someone crashing  anyway...

lcgamboa, 2018-01-09:
The PC wraps around to zero when the flash memory end it is the behavior of real AVR microcontroller.  
I have problems when the reset_pc is set to bootloader address but the code loaded in flash don't have the bootloader code.  In real ATMEGA it's work, in simavr don't.
Of course if the flashend is a power of two value a mask will be a best solution like: 

``
avr_flashaddr_t new_pc = (avr->pc + 2 )& 0x0FFFF; // future "default" pc
``

Best regards.

buserror, 2018-01-09:
Well there is another method which I would prefer is for example to resize the flash to size+2, and add an invalid opcode there -- for every AVR. The invalid opcode handler in the core would wrap the PC, but *also* be able to log the condition (which is quite unusual) with zero performance impact.

buserror, 2018-01-10:
Well I've pushed a patch that implements exactly what I said. it's untested but should do the job. please give feedback...

lcgamboa, 2018-01-11:
Your solution is better because it does not overload the simulator. 

But there was apparently something missing in the implementation. Here is the error while running with your solution:

```
avr_sadly_crashed
r0=00 r1=00 r2=00 r3=00 r4=00 r5=00 r6=00 r7=00 r8=00 r9=00 r10=00 r11=00 r12=00 r13=00 r14=00 r15=00
r16=00 r17=00 r18=00 r19=00 r20=00 r21=00 r22=00 r23=00 r24=00 r25=00 XL=00 XH=00 YL=00 YH=00 ZL=00 ZH=00 
Y+00=00 Y+01=00 Y+02=00 Y+03=00 Y+04=00 Y+05=00 Y+06=00 Y+07=00 Y+08=00 Y+09=00 Y+10=00 Y+11=00 Y+12=00 Y+13=00 Y+14=00 Y+15=00 Y+16=00 Y+17=00 Y+18=00 Y+19=00 
*** CYCLE 2048PC 8000
*** 0000: unknown                   RESET -1; sp 0000
*** 0000: unknown                   RESET -2; sp 0000
*** 0000: unknown                   RESET -3; sp 0000
*** 0000: unknown                   RESET -4; sp 0000
*** 0000: unknown                   RESET -5; sp 0000
*** 0000: unknown                   RESET -6; sp 0000
*** 0000: unknown                   RESET -7; sp 0000
*** 0000: unknown                   RESET -8; sp 0000
*** 0000: unknown                   RESET -9; sp 0000
*** 0000: unknown                   RESET -10; sp 0000
*** 0000: unknown                   RESET -11; sp 0000
*** 0000: unknown                   RESET -12; sp 0000
*** 0000: unknown                   RESET -13; sp 0000
*** 0000: unknown                   RESET -14; sp 0000
*** 0000: unknown                   RESET -15; sp 0000
*** 0000: unknown                   RESET -16; sp 0000
*** 0000: unknown                   RESET -17; sp 0000
*** 0000: unknown                   RESET -18; sp 0000
*** 0000: unknown                   RESET -19; sp 0000
*** 0000: unknown                   RESET -20; sp 0000
*** 0000: unknown                   RESET -21; sp 0000
*** 0000: unknown                   RESET -22; sp 0000
*** 0000: unknown                   RESET -23; sp 0000
*** 0000: unknown                   RESET -24; sp 0000
*** 0000: unknown                   RESET -25; sp 0000
*** 0000: unknown                   RESET -26; sp 0000
*** 0000: unknown                   RESET -27; sp 0000
*** 0000: unknown                   RESET -28; sp 0000
*** 0000: unknown                   RESET -29; sp 0000
*** 0000: unknown                   RESET -30; sp 0000
*** 0000: unknown                   RESET -31; sp 0000
Stack Ptr 08ff/08ff = 0 
```

lcgamboa, 2018-01-12:
The problem is the call to function crash on code below.  Without the call the modified code works.

https://github.com/buserror/simavr/blob/1b6e6ed10a92438da47fe99864698db9ac2169b6/simavr/sim/sim_core.c#L635-L642

buserror, 2018-01-16:
The 'proper' fix is to change the avr->flashend to avr->flashend + 2 -- I'll commit that later...

lcgamboa, 2018-01-16:
Your proposed solution (change the avr->flashend to avr->flashend + 2) did not work for me.

The best solution is replace the crash code:

https://github.com/buserror/simavr/blob/6660aba00a898ff4db6c689e35e7fd03b4f9d47d/simavr/sim/sim_core.c#L638-L645
by:
```
	if (unlikely(avr->pc >= avr->flashend)) {
	      avr->pc =  avr->pc   % (avr->flashend+1);
	}

	uint32_t                opcode = _avr_flash_read16le(avr, avr->pc);
	avr_flashaddr_t	        new_pc = avr->pc + 2 ;// future "default" pc
```

closed 2020-04-11 by lcgamboa