buserror/simavr #270: Simduino seems to choke on Serial output

idoc, 2018-01-18:
I've been trying to use simduino to test some Arduino code, and I just can't get any of my code to run for more than a few seconds. After much experimentation, it seems to be serial writes that make the code hang.

My simplest test program is
```
#include <Arduino.h>

int i = 0;
void setup() {
    delay(2000);
    Serial.begin(115200);
}

void loop() {
    Serial.println(i++);
}
```
When I run this program in simduino (using SIMAVR_UART_XTERM=1 to capture the output), it hangs after printing 222 lines.

Adding some delays between the prints, or a Serial.flush() after every one of them, will get the program to run a little longer, but if will still hang pretty quickly.

The same program runs smoothly on an actual Arduino.

I'm running the latest code from the master branch on OS X 10.13.2. Any help would be appreciated.

Thanks!

lcgamboa, 2018-01-26:
I have problems with simduino uart_pty too, the solution proposed in the link below solve the problem:
https://groups.google.com/forum/#!topic/simavr/eSVV90dFh8U
Maybe it's solve your problem.

idoc, 2018-02-04:
Thanks, @lcgamboa! That solved my UART problem.

closed 2018-02-04 by idoc

TomasHubelbauer, 2019-08-22:
Hey @idoc. Would you me writing up a short guide on how to go from having an Arduino sketch file to running it in simavr and seeing its serial output in your terminal? This is a bit way over my head right now and I have not been able to figure it out based on the readme but I see you know how to do it as you mention a sketch source code in your issue while at the same time being able to run it expecting to see the values of serial writes in the output.

I assume I need to compile the sketch first (is this what avrdude is for?) and then take that hex file and construct a command which tells simavr to use the right core and the resulting hex file. But I don't know how to do it. :-) Any help would be appreciated.