buserror/simavr #285: _MMCU_ instructions not working (avg-g++)?

larsks, 2018-05-17:
I'm using the [Arduino-Makefile][] project to build a simple program for an attiny85. I wanted to try running this via `simavr`. I tried adding the following to my code:

    const struct avr_mmcu_vcd_trace_t _mytrace[]  _MMCU_ = {
      { AVR_MCU_VCD_SYMBOL("PORTB"), .what = (void*)&PORTB, },
    };

When trying to compile the code, I get:

    sample.ino:18:1: error: C99 designator 'name' outside aggregate initializer
     };
     ^
    sample.ino:18:1: sorry, unimplemented: non-trivial designated initializers not supported

The compile command line looks like this:

    avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -I$HOME/src/simavr/ -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=185 -DARDUINO_ARCH_AVR -I$HOME/lib/arduino/hardware/arduino/avr/cores/arduino -I$HOME/.arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8     -I$HOME/Arduino/libraries/DallasTemperature   -I$HOME/Arduino/libraries/OneWire -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always sample.ino -o build-ATtinyX5-attiny85/sample.ino.o

Will the tracing feature work with avg-g++?

larsks, 2018-05-18:
```
$ avr-g++ --version
avr-g++ (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hovercraft-github, 2018-05-18:
@larsks  AFAIK, you have to place any _MMCU_ sections into .c file and compile it with avr-gcc.
This is related to c++ structure/class initialization limitations.
Also you can have c++ files in your project too. Preferably, link everything using avr-g++ and remember to add the following linker option: -Wl,--undefined=_mmcu,--section-start=.mmcu=0x910000

larsks, 2018-05-18:
Thanks.

So, putting the simavr directives into a `.c` file (`metadata.c`), it gets compiled (successfully!) into an object file like this:

```
avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ \
  [...a bunch of -I directives...]
  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=184 -DARDUINO_ARCH_AVR \
  -Wall -ffunction-sections -fdata-sections \
  -Os -std=gnu11 -flto -fno-fat-lto-objects \
  -fdiagnostics-color=always metadata.c \
  -o build-ATtinyX5-attiny85/metadata.c.o
```

But the resulting object file has no `.mmcu` section:

```
$ objdump -h build-ATtinyX5-attiny85/metadata.c.o
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000000  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  00000034  2**0
                  ALLOC
  3 .gnu.lto_.profile.45131e2142b701f9 0000000f  00000000  00000000  00000034  2**0
                  CONTENTS, READONLY, EXCLUDE
  4 .gnu.lto_.inline.45131e2142b701f9 0000000e  00000000  00000000  00000043  2**0
                  CONTENTS, READONLY, EXCLUDE
  5 .gnu.lto_.pureconst.45131e2142b701f9 0000000f  00000000  00000000  00000051  2**0
                  CONTENTS, READONLY, EXCLUDE
  6 .gnu.lto_.symbol_nodes.45131e2142b701f9 00000017  00000000  00000000  00000060  2**0
                  CONTENTS, READONLY, EXCLUDE
  7 .gnu.lto_.refs.45131e2142b701f9 0000000f  00000000  00000000  00000077  2**0
                  CONTENTS, READONLY, EXCLUDE
  8 .gnu.lto_.decls.45131e2142b701f9 00000477  00000000  00000000  00000086  2**0
                  CONTENTS, READONLY, EXCLUDE
  9 .gnu.lto_.symtab.45131e2142b701f9 00000018  00000000  00000000  000004fd  2**0
                  CONTENTS, READONLY, EXCLUDE
 10 .gnu.lto_.opts 000000a3  00000000  00000000  00000515  2**0
                  CONTENTS, READONLY, EXCLUDE
 11 .comment      00000012  00000000  00000000  000005b8  2**0
                  CONTENTS, READONLY
```

And `metadata.c` is as above:

```c
#include <Arduino.h>
#include <avr_mcu_section.h>

const struct avr_mmcu_vcd_trace_t _mytrace[]  _MMCU_ = {
    { AVR_MCU_VCD_SYMBOL("PORTB"), .what = (void*)&PORTB, },
};
```
Should I expect to see the `.mmcu` section in the object file?

larsks, 2018-05-19:
That fixed the `.o` file...

...but it looks as if [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile/) links the final `.elf` file with `-Wl,--gc-sections`, which of course dropped the `.mmcu` section from the final object.  I've modified that as well, and now I'm able to generate traces as expected.

Thanks for your help!

closed 2018-05-19 by larsks

msquirogac, 2020-05-23:
Sorry to bring this back, but these are some of my observations.
Certainly, as @hovercraft-github suggested, the **flto** flag drops all the unused symbols and a possible workaround is to remove said flag.
However, I think a better solution would be to force the compiler/linker to add the symbols by adding  the **unused** attribute, like explained [here](https://stackoverflow.com/questions/31637626/whats-the-usecase-of-gccs-used-attribute)
That way we can be sure that the symbols are always included even if the **flto** flag is used.
I've tested it and the .mmcu section doesn't get dropped anymore.