buserror/simavr #286: USB Support Revisited - VHCI

aznoohwee, 2018-05-18:
I recently was able to get the usb example to work again and I will be sharing details in my fork of the repository soon.

Essentially it came down to updating the VHCI kernel modules and projects to support more recent kernels.

So far I've tested on Ubuntu 18.04 and Fedora 27 and have been able to test firmwares with raw hid and serial interfaces.

leo60228, 2018-07-13:
Where is this kernel module?

aznoohwee, 2018-07-14:
https://sourceforge.net/projects/usb-vhci/?source=navbar

It requires a few tweaks to get working with more recent kernels.  I’m working on getting the change set contributed back at sourceforge.

leo60228, 2018-07-14:
Is this fork publically available?

aznoohwee, 2018-07-14:
https://sourceforge.net/u/aznoohwee/usb-vhci/ci/master/tree/

Can you access this?

aznoohwee, 2018-07-14:
https://github.com/aznoohwee/simavr/commit/5ced648b063778ae162ce1d25d1fa6828f3fdd9a

And this will get you going.

aznoohwee, 2018-07-14:
```
From 61fa1910ddaca4c937fb31b81117dbf1f9be7e28 Mon Sep 17 00:00:00 2001
From: Tam Do <tdo.pro@gmail.com>
Date: Sat, 14 Apr 2018 08:48:35 -0500
Subject: [PATCH] add missing header file (new for linux kernels ~4.2.18 ish)

---
 usb-vhci-iocifc.c | 1 +
 1 file changed, 1 insertion(+)

diff --git a/usb-vhci-iocifc.c b/usb-vhci-iocifc.c
index f09d51f..2261f56 100644
--- a/usb-vhci-iocifc.c
+++ b/usb-vhci-iocifc.c
@@ -32,6 +32,7 @@
 #include <linux/wait.h>
 #include <linux/list.h>
 #include <linux/platform_device.h>
+#include <linux/uaccess.h>
 #include <linux/usb.h>
 #include <linux/fs.h>
 
-- 
2.14.1
```

This is for the usb-vhci project.  Sorry am having issues with source forge.

leo60228, 2018-07-15:
Thanks! Does there not need to be a (ex.) 4.11 header in `linux/`?

aznoohwee, 2018-07-15:
Is this what you mean?

```
diff --git a/usb-vhci-iocifc.c b/usb-vhci-iocifc.c
index f09d51f..3c3df7b 100644
--- a/usb-vhci-iocifc.c
+++ b/usb-vhci-iocifc.c
@@ -32,6 +32,12 @@
 #include <linux/wait.h>
 #include <linux/list.h>
 #include <linux/platform_device.h>
+#include <linux/version.h>
+
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 2, 18)
+#include <linux/uaccess.h>
+#endif
+
 #include <linux/usb.h>
 #include <linux/fs.h>
```

leo60228, 2018-07-15:
There's a directory named Linux in the root of the source code. Does that
not need to be changed?

aznoohwee, 2018-07-16:
Ahh never paid any attention to those.  Just install the headers for your system via your package management system.

```
sudo apt-get install linux-headers-$(uname -r)
```

or for fedora/centos
```
yum -y install kernel-devel kernel-headers
```

vintagepc, 2021-02-11:
Worth a shot poking here but I doubt I'll get anything since it's 2+ years old...

Anyway, I've been experimenting trying to get USB working and though the kernel module and libusb are compiled, USB seems very... unreliable. In other words, it's not consistent and sometimes I just get a wall of NAKs when starting the sim, other times it kinda sorta works but the transmission is bad (i.e. device descriptors are truncated). This smells of either a threading issue or a timing issue and I'm curious whether this has been observed before.