buserror/simavr #292: Advice about ssd1306 images

stuveri, 2018-06-06:
Hi,
More support than an issue - but, I have been unable to make any of the hex arrays out there on the web or ones that I've made myself render anything like they're supposed to when I replace the code in the images.c file, recompile and run. The SIMAVR graphic runs fine. Could you please advise if there is a tool to allow me to generate graphic hex arrays that would be compatible. I've tried a lot of different ways but nothing is working. Thanks in advance for any help.

marcidy, 2018-06-06:
Have you compared with the datasheet for the proper format?

As there is a working example, I would expect you can start fiddling with
one byte at a time to get results.  Have you tried that?

Datasheet will be your only instructions.

I've never used it but that's what I would read first before trying to use
random other code to understand the working example.

stuveri, 2018-06-06:
I've tried tweaking the hex within the array and am at a loss so far to identify the pattern.
It's frustrating as a working example would usually be enough to go on and there 
are plenty of people happily generating graphics for their displays however their
hex arrays do not seem to work with simavr.  When I drop in other arrays that allegedly
work nothing is being displayed for the image within the emulation until it moves on to
the lines/bitwise graphics later in the demo. I'll keep at it.

stuveri, 2018-06-07:
I'll answer my own query - run length encoding - hadn't encountered it before but
that's the default method used to squirt bits into the memory buffer in ssd1306_write_image_fb().
Should have paid more attention to the comments!

closed 2018-06-07 by stuveri