buserror/simavr #296: SPI slave ISR not raised

aznoohwee, 2018-07-03:
I found that IRQ is not raised when simulating firmware operating as a SPI slave.

PR is WIP to see if there is anything else that needs to be done but here is a quick fix.

static avr_cycle_count_t avr_spi_raise(struct avr_t * avr, avr_cycle_count_t when, void * param)
{
	avr_spi_t * p = (avr_spi_t *)param;
	
	if (avr_regbit_get(avr, p->spe)) {
		// in master mode, any byte is sent as it comes..
		if (avr_regbit_get(avr, p->mstr)) {
			avr_raise_interrupt(avr, &p->spi);
			avr_raise_irq(p->io.irq + SPI_IRQ_OUTPUT, avr->data[p->r_spdr]);
		}
        // in slave mode, bytes are sent but no interrupt is raised
        // TODO check for state of SS line??
        else {
			avr_raise_irq(p->io.irq + SPI_IRQ_OUTPUT, avr->data[p->r_spdr]);
        }
	}
	return 0;
}

aznoohwee, 2018-07-05:
Related to #222

buserror, 2018-07-05:
Good catch -- to be fair I've haven't had to use spi slave in a long time, so that probably had been left as 'an exercise to the reader' ;-)
Please PR, even if it's not all polished, as long as you are happy it works...

closed 2018-07-18 by aznoohwee