buserror/simavr #297: Backtrace stopped: Cannot access memory at address 0x800900

zhujiu, 2018-07-05:
hi, all,
 
[image]

as you can see, i had connected succed to the simavr, but when I use bt ,it shows this issue, i do not know why, please help  me , i am really frustrated.....

[image]

ccrause, 2018-07-05:
This is strange, 0x900 is 1 byte past the end of RAM (0x8FF), and reading two bytes past the end of RAM is allowed by this code:
https://github.com/buserror/simavr/blob/51de22447b70aff741c5bd7c1759db8d343e5682/simavr/sim/sim_gdb.c#L371

Perhaps enable display of the remote serial protocol in gdb (set debug remote 1) before calling backtrace, this way one can see what memory gdb is trying to read in simavr.

Also, did you include debug information in the compiled program and loaded atmega328p_dummy_blinky.elf in gdb using the file command?  This way gdb is better informed on where in memory data is stored.

buserror, 2018-07-05:
I just think his stack is trashed, and the backtrace fails because of that; #0 /should/ be the bottom of the stack anyway, so trying to find a bug in gdb handler isn't going to help one bit.

ccrause, 2018-07-06:
I checked the backtrace over debugwire at start (tiny45):
```
(gdb) file blink.elf 
Reading symbols from blink.elf...done.
(gdb) target remote :2345
Remote debugging using :2345
0x00000000 in _START ()
(gdb) bt
#0  0x00000000 in _START ()
(gdb) 
```

Compared to simavr backtrace:
```
(gdb) file blink.elf
(gdb) target remote :1234
Remote debugging using :1234
0x00000000 in _START ()
(gdb) bt
#0  0x00000000 in _START ()
Backtrace stopped: Cannot access memory at address 0x800160
(gdb) 
```
It seems as if simavr is returning an error and not two zero bytes as expected from the code. Perhaps ram_end is off by 1?

ccrause, 2018-07-06:
So after comparing different scenarios it seems as if gdb will do the following when the stack pointer is pointing to end of RAM, such as when the program hasn't started yet:

- If the server responded with a memory map (qXfer:memory-map:read) as simavr does then gdb will display the original message "Backtrace stopped: Cannot access memory..." and a memory read will not be requested by gdb.
- If the server doesn't support the qXfer:memory-map request, gdb will try and read a couple of bytes past the end of RAM to try and construct a backtrace (because gdb doesn't know which MCU it is debugging).
- If gdb has loaded a file with debug info, it will read the next two bytes past end of RAM.

So the observed message in gdb is a peculiarity of gdb and not an error in simavr.

zhujiu, 2018-07-07:
@ccrause 
[image]
this is my command