buserror/simavr #303: segfault when running simduino.elf

JohnPeng47, 2018-08-15:
Running command below results in a segfault:
run_avr -m atmega328 --freq 1400000 -g obj-x86_64-linux-gnu/simduino.elf

Core dump from gdb (pwngdb plugin so everything looks a bit different):

   333 		GElf_Shdr shdr;                 /* Section Header */
   334 		gelf_getshdr(scn, &shdr);
   335 		char * name = elf_strptr(elf, elf_header.e_shstrndx, shdr.sh_name);
   336 		printf("Walking elf section '%s'\n", name);
   337 
 ► 338 		if (!strcmp(name, ".text"))
   339 			data_text = elf_getdata(scn, NULL);
   340 		else if (!strcmp(name, ".data"))
   341 			data_data = elf_getdata(scn, NULL);
   342 		else if (!strcmp(name, ".eeprom"))
   343 			data_ee = elf_getdata(scn, NULL);
Stack trace
 ► f 0     7ffff7b080c4 elf_read_firmware+356
   f 1           40195e main+2926
   f 2     7ffff7538830 __libc_start_main+240
Program received signal SIGSEGV (fault address 0x0)
pwndbg> print name
$1 = 0x0

Appears to be null pointer deference by name, which implies that elf_strptr returned NULL. Any ideas?

buserror, 2018-08-15:
Well without any idea of the OS and toolchain version you are using, what do you think we can do?

JohnPeng47, 2018-08-15:
Ubuntu 16.0.4 and output from avr-gcc -v:
Using built-in specs.
Reading specs from /usr/lib/gcc/avr/4.9.2/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/4.9.2/lto-wrapper
Target: avr
Configured with: ../src/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=avr CFLAGS='-g -O2 -fstack-protector-strong -Wformat ' CPPFLAGS=-D_FORTIFY_SOURCE=2 CXXFLAGS='-g -O2 -fstack-protector-strong -Wformat ' FCFLAGS='-g -O2 -fstack-protector-strong' FFLAGS='-g -O2 -fstack-protector-strong' GCJFLAGS='-g -O2 -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fstack-protector-strong -Wformat ' OBJCXXFLAGS='-g -O2 -fstack-protector-strong -Wformat '
Thread model: single
gcc version 4.9.2 (GCC)

evandrix, 2019-12-29:
*bump* i'm facing the exact same issue