buserror/simavr #313: Attiny10 support

girst, 2018-12-19:
Hi, I'd like to be able to simulate an Attiny4/5/9/10 for a project, but this core is not currently supported by simavr (or any other FOSS simulator). A [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8127-AVR-8-bit-Microcontroller-ATtiny4-ATtiny5-ATtiny9-ATtiny10_Datasheet.pdf) is available. 

I haven't fully grokked the core definitions, otherwise I'd done it myself ;-) (and of course, no grudges if you don't implement it)

leg0, 2019-01-16:
I am working on attiny9 support (https://github.com/leg0/simavr/tree/add_tiny9). Work is in progress, but maybe it helps.

girst, 2019-01-17:
attiny4/9 support is more than enough for my use case[1]. I'll have a 
look, thanks!

[1]: https://gir.st/chiptunes.htm

0x5c, 2019-02-23:
As-there been any progress since? I would be interested in it, since it seems to be the only way to simulate an ATTiny10.

leg0, 2019-02-27:
> As-there been any progress since? I would be interested in it, since it seems to be the only way to simulate an ATTiny10.

Not much progress. It kind of works at the moment. Depends on what you do. CCP register is definitely broken at the moment. Some modes of timer also work.

If you're ok with running Windows, then this is not the only way to simulate an ATtiny10 - Atmel Studio simulates it.