buserror/simavr #315: avr ld: -pie not supported

Petross404, 2018-12-22:
I cloned the repo on my Gentoo box and I am trying to compile simavr. I am stuck at this error.
```
make 
make -C simavr RELEASE=0
make[1]: Entering directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
CONF obj-x86_64-pc-linux-gnu/cores.deps
make obj config
make[2]: Entering directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
make[2]: Nothing to be done for 'obj'.
CONF sim_core_decl.h
make[2]: Leaving directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
make libsimavr run_avr
make[2]: Entering directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
CC sim/sim_interrupts.c
CC sim/sim_utils.c
CC sim/sim_gdb.c
CC sim/sim_cycle_timers.c
CC sim/sim_core.c
CC sim/sim_elf.c
CC sim/sim_cmds.c
CC sim/sim_irq.c
CC sim/sim_hex.c
CC sim/sim_avr.c
CC sim/sim_vcd_file.c
CC sim/sim_io.c
CC sim/avr_lin.c
CC sim/avr_flash.c
CC sim/avr_twi.c
CC sim/avr_adc.c
CC sim/avr_spi.c
CC sim/avr_usb.c
CC sim/avr_ioport.c
CC sim/avr_uart.c
CC sim/avr_extint.c
CC sim/avr_acomp.c
CC sim/avr_eeprom.c
CC sim/avr_timer.c
CC sim/avr_bitbang.c
CC sim/avr_watchdog.c
CORE cores/sim_mega1280.c
CORE cores/sim_mega1281.c
CORE cores/sim_mega1284.c
CORE cores/sim_mega128.c
CORE cores/sim_mega128rfa1.c
CORE cores/sim_mega128rfr2.c
CORE cores/sim_mega164.c
CORE cores/sim_mega168.c
CORE cores/sim_mega169.c
CORE cores/sim_mega16.c
CORE cores/sim_mega16m1.c
CORE cores/sim_mega2560.c
CORE cores/sim_mega324a.c
CORE cores/sim_mega324.c
CORE cores/sim_mega328.c
CORE cores/sim_mega32.c
CORE cores/sim_mega32u4.c
CORE cores/sim_mega48.c
CORE cores/sim_mega644.c
CORE cores/sim_mega88.c
CORE cores/sim_mega8.c
CORE cores/sim_megax4.c
CORE cores/sim_megax8.c
CORE cores/sim_megax.c
CORE cores/sim_megaxm1.c
CORE cores/sim_tiny13.c
CORE cores/sim_tiny2313a.c
CORE cores/sim_tiny2313.c
CORE cores/sim_tiny24.c
CORE cores/sim_tiny25.c
CORE cores/sim_tiny4313.c
CORE cores/sim_tiny44.c
CORE cores/sim_tiny45.c
CORE cores/sim_tiny84.c
CORE cores/sim_tiny85.c
CORE cores/sim_tinyx4.c
CORE cores/sim_tinyx5.c
CORE cores/sim_usb162.c
AR obj-x86_64-pc-linux-gnu/libsimavr.a
SHARED obj-x86_64-pc-linux-gnu/libsimavr.so.1
ln -sf libsimavr.so.1 obj-x86_64-pc-linux-gnu/libsimavr.so
CC sim/run_avr.c
LD obj-x86_64-pc-linux-gnu/run_avr.elf
ln -sf obj-x86_64-pc-linux-gnu/run_avr.elf run_avr
make[2]: Leaving directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
make[1]: Leaving directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/simavr'
make -C tests RELEASE=0
make[1]: Entering directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/tests'
AVR-CC atmega48_disabled_timer.c
cc1: warning: -fPIC is not supported [-fPIC]
cc1: warning: -fPIE is not supported [-fPIE]
cc1: warning: -fstack-protector not supported for this target
/usr/libexec/gcc/avr/ld: -pie not supported
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile.common:160: atmega48_disabled_timer.axf] Error 1
make[1]: Leaving directory '/mnt/Volume_3/Gentoo/projects/Ebuilds/simavr/tests'
make: *** [Makefile:21: build-tests] Error 2
```

Any ideas?

buserror, 2019-02-25:
make V=1 and paste the output. Seems your box tries to add flags that aren't supposed to be there, and avr-gcc doesn't like them...

Petross404, 2019-03-02:
[simavr.log](https://github.com/buserror/simavr/files/2922171/simavr.log)
That's the output. I think that my gentoo profile forces some hardening options on GCC.

buserror, 2019-03-19:
Well these options are related to x86 gcc, not avr-gcc, so it's certainly not a bug of simavr!

closed 2019-03-19 by buserror

Petross404, 2019-07-21:
> Well these options are related to x86 gcc, not avr-gcc, so it's certainly not a bug of simavr!

Still, isn't the x86 gcc that is compiling simavr?

buserror, 2019-08-15:
Not all, the AVR-CC lines means it's trying to compile the examples (which are AVR code) using obviously -fpie from a default x86 CFLAGS line

Petross404, 2019-08-17:
> Not all, the AVR-CC lines means it's trying to compile the examples (which are AVR code) using obviously -fpie from a default x86 CFLAGS line

So should I force `make` to use `avr-gcc`?