buserror/simavr #326: ATTiny USI (universal serial interface) peripheral unimplemented

JarrettBillingsley, 2019-03-23:
Spent a few hours trying to figure out what I was doing wrong with my code. #210 hinted at the problem but ultimately: there's no USI support. 

Granted, this is not listed in the README.md on the front page, so shame on me for not reading it... but either it should be mentioned as unimplemented, or an error should be issued if you try to use the peripheral (until it is actually implemented). 

I might try implementing it, but I don't know if I have the time.

rvodden, 2019-05-08:
I'm happy to have a crack at this if you've not started yet @jarretbillingsley. Similarly if you've got WIP code, I'm more than happy to help!

JarrettBillingsley, 2019-05-08:
Ah yes sorry! I got started on this and didn't quite finish. I've got enough of it working to support simple SPI and I've been using it to test a project. Let me see if I can get this repo forked properly.

JarrettBillingsley, 2019-05-08:
Boom. @rvodden https://github.com/JarrettBillingsley/simavr/tree/tinyusi

gatk555, 2025-03-31:
Presumed fixwd by PR #514.

closed 2025-04-02 by gatk555