buserror/simavr #327: Having Error "No names found, cannot describe anything" While Installing

xiaofuhu, 2019-03-23:
Here is the part of output where the error happens. Please help!
```
==> make all HOMEBREW_PREFIX=/usr/local RELEASE=1
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr RELEASE=1
fatal: No names found, cannot describe anything.
../Makefile.common:56: *** Please install avr-gcc: port install avr-gcc avr-libc.  Stop.
make: *** [build-simavr] Error 2
```

JarrettBillingsley, 2019-03-30:
Seems like you incorrectly pulled the repository. That "no names found" error is from the `git describe` command. Try to use `git clone` to get the simavr repository instead.

Also you need to install `avr-gcc`.

gatk555, 2025-04-08:
Answered, no response: closing.

closed 2025-04-08 by gatk555