buserror/simavr #330: examples/* doesn't create ${OBJ} directory automatically

marcidy, 2019-03-30:
The directory "obj-x86_64-linux-gnu" needs to be manually created to get the examples to compile.  Without it, I get this error (for i2ctest for example):
`i2ctest.c:87:1: fatal error: opening dependency file obj-x86_64-linux-gnu/i2ctest.d: No such file or directory`

Clearly not a big deal but it's annoying if someone new wants to just run them, and it's a nuance of compiling to find it and fix it. 

The directories don't ship since `.gitignore` has `obj-*`, and everything else is handled in the `simavr/simavr/Makefile` which creates the obj- directory on line 137:
`     mkdir -p ${OBJ} ; echo >${OBJ}/cores.deps ; echo >$(DEBUGLOG) ;\`

Each example Makefile has 
`all: obj ... `
where obj is defined in Makefile.common, and I think correctly, but I'm not clear why it's not inheriting.  

I changed this to
`all: ${OBJ} ...`
And this necessitated moving all the targets below the `include ../../Makefile.common` in each Makefile.

I tried other fixes but they caused more trouble than they were worth imo.  Ideally the `obj` target will inherit correctly but I can't seem to figure that one out.

`$ make` works at each level. 
Tested on debian 9, i dont have a mac to test on.

I have the PR ready if interested but I'm not permissioned on the repo.

hinxx, 2020-08-28:
> Again im not sure why using the ${OBJ} target in obj: ${OBJ} does not work.

It is because of `VPATH += ../parts` that is found in the `Makefile` of the examples (I stumbled upon this today with `simavr/examples/board_simduino/Makefile`) and got really annoyed of this behavior as well. AFAICT, the `parts/obj-x86_64-linux-gnu` exists and makes the `simavr/examples/board_simduino/Makefile` think it does not need to create `obj-x86_64-linux-gnu` in the current folder.

Reading http://make.mad-scientist.net/papers/rules-of-makefiles/#rule3 and http://make.mad-scientist.net/papers/how-not-to-use-vpath/ suggests that the  use case for `VPATH` in `simavr` Makefile seems to be not proper causing this kind of behavior.

buserror, 2020-08-31:
Well VPATH works file, we just need to add a few 'optional targets' wiht | (pipe) to add the OBJ dependencies to some of the rules. I'll have to revisit that.

gatk555, 2026-05-09:
Suggested fix commited: c02488e
.