buserror/simavr #336: Suggestion: implement the xmega and new attiny parts

wjl, 2019-04-13:
You might consider implementing the xmega and newer tinyavr architectures so that those parts could be simulated. I use both of these parts quite a bit and it would be awesome to see them supported in simavr.

The newer architectures only add a few instructions, which would probably be pretty easy to add. There are also some peripheral changes, but those could always be added later.

MaGaroo, 2025-02-26:
I already know some people in academic communities that would require atxmega support. However, I think the implementation would introduce a lot of changes to be covered in a single pull request. So, if we want to support it, we should break it into smaller tasks and approach it step by step.

@buserror I would like to know your opinion about it. If you agree, I can work on it in my free time.

</hr>

This weekend, I spent some time on it. There were some required changes that I could think of. Of course, there might be a lot more to do. I'll list them here, hoping that it will help the person going for the implementation. (It might be me as well!)

### IO Ports
Currently, the max value of an IO Port's address is hardcoded as `280` in the source code (`MAX_IOs` in `sim_avr.h:70`). It should be changed since e.g. in atxmega128d4, we have such a line:
```c
#define PORTB_OUT  _SFR_MEM8(0x0624)
```
Also, each part of an IO port occupies 4 bytes in the address space:
```c
#define PORTB_DIR  _SFR_MEM8(0x0620)
#define PORTB_DIRSET  _SFR_MEM8(0x0621)
#define PORTB_DIRCLR  _SFR_MEM8(0x0622)
#define PORTB_DIRTGL  _SFR_MEM8(0x0623)

#define PORTB_OUT  _SFR_MEM8(0x0624)
#define PORTB_OUTSET  _SFR_MEM8(0x0625)
#define PORTB_OUTCLR  _SFR_MEM8(0x0626)
#define PORTB_OUTTGL  _SFR_MEM8(0x0627)
```
Which is different from the traditional way: (example from atmega128)
```c
/* Data Direction Register, Port B */
#define DDRB      _SFR_IO8(0x17)

/* Data Register, Port B */
#define PORTB     _SFR_IO8(0x18)
```

There are as well more details to it. It seems the IO ports can be remapped, but I didn't dig deeper into the details. All of these prove that a whole new struct should be used for IO ports instead of the current `struct avr_io_t`

### EEPROM
The EEPROM was memory-mapped in the old architecture. In the new architecture, it works with NVM interface as well. So, a new module has to be defined to handle (simulate) NVM interface operations.

### Instruction Set
There are some new instructions in the new architecture. Here's a summary I got from ChatGPT:

> The XMEGA series introduced some new instructions to improve performance and efficiency, such as:
> 
> ELPM Rn, Z → Improved extended load from flash
> MOVW Rdh, Rrl → Optimized word move
> LD/ST with pre/post increment → Faster memory access
> DES → Built-in hardware DES encryption instruction

The simulation of these instructions has to be implemented as well.

ccrause, 2025-02-26:
Another detail to consider is that the clock cycles for a few instructions are different between the different variants (e.g. CALL, LD, LDD, LDS, ST etc.).

gatk555, 2025-03-31:
Hello @MaGaroo,

I only just saw this, so a rather late response.  Although @buserror did not respond to your prompt, he has granted me the power to integrate PRs and I would like to see this happen.  I think I can contribute a simplified interrupt handler, intended as a base to implement the additional priority levels and round-robin stuff, and might even complete it, if you do GPIO.  I might also look at the ADC and ACs for the newer chips.  More later: dinner gong!