buserror/simavr #339: Wrong ADC definition at-tiny-24

djfd, 2019-04-15:
Hi,

playing with my firmware I noticed that [ADC definition of tiny24](https://github.com/buserror/simavr/blob/master/simavr/cores/sim_tinyx4.h#L150) is in contradiction with [at-tiny-24a datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/doc8183.pdf) (pp.145-146)

Currently [only 4 pins of ```ADMUX``` are used](https://github.com/buserror/simavr/blob/master/simavr/cores/sim_tinyx4.h#L116), while datasheet uses all six bits, and consequently modes themselves described not in full and simply wrong.

Well, the software model files are for models without ```A``` suffix, so I was curious if ```24A``` version differs from pure ```24```, and compared it with [at-tiny-24 datasheet](http://ww1.microchip.com/downloads/en/devicedoc/Atmel-7701_Automotive-Microcontrollers-ATtiny24-44-84_Datasheet.pdf) briefly. It seems there are not any differences. Or am I wrong, or maybe missing something?

Proposed fix (as a patch, pls do not ask for PR, sorry), which works for me (I was able to read the chip temperature sensor)

```patch
--- a/simavr/simavr/cores/sim_tinyx4.h	2019-04-15 21:54:36.054957167 +1000
+++ b/simavr/simavr/cores/sim_tinyx4.h	2019-04-15 21:53:54.761157331 +1000
@@ -116,7 +116,8 @@
     .adc = {
         .r_admux = ADMUX,
         .mux = { AVR_IO_REGBIT(ADMUX, MUX0), AVR_IO_REGBIT(ADMUX, MUX1),
-                    AVR_IO_REGBIT(ADMUX, MUX2), AVR_IO_REGBIT(ADMUX, MUX3),},
+                    AVR_IO_REGBIT(ADMUX, MUX2), AVR_IO_REGBIT(ADMUX, MUX3),
+                    AVR_IO_REGBIT(ADMUX, MUX4), AVR_IO_REGBIT(ADMUX, MUX5),},
         .ref = { AVR_IO_REGBIT(ADMUX, REFS0), AVR_IO_REGBIT(ADMUX, REFS1), },
         .ref_values = {
                 [0] = ADC_VREF_VCC, [1] = ADC_VREF_AREF,
@@ -147,17 +148,43 @@
         },
         .bin = AVR_IO_REGBIT(ADCSRB, BIN),
 
-        .muxmode = {
+        .muxmode = { /* 64 entries, including offset calibration */
             [0] = AVR_ADC_SINGLE(0), [1] = AVR_ADC_SINGLE(1),
             [2] = AVR_ADC_SINGLE(2), [3] = AVR_ADC_SINGLE(3),
+            [4] = AVR_ADC_SINGLE(4), [5] = AVR_ADC_SINGLE(5),
+            [6] = AVR_ADC_SINGLE(6), [7] = AVR_ADC_SINGLE(7),
 
-            [ 4] = AVR_ADC_DIFF(2, 2,   1), [ 5] = AVR_ADC_DIFF(2, 2,  20),
-            [ 6] = AVR_ADC_DIFF(2, 3,   1), [ 7] = AVR_ADC_DIFF(2, 3,  20),
-            [ 8] = AVR_ADC_DIFF(0, 0,   1), [ 9] = AVR_ADC_DIFF(0, 0,  20),
-            [10] = AVR_ADC_DIFF(0, 1,   1), [11] = AVR_ADC_DIFF(0, 1,  20),
-            [12] = AVR_ADC_REF(1100),    // Vbg
-            [13] = AVR_ADC_REF(0),        // GND
-            [15] = AVR_ADC_TEMP(),
+            /* no ofs.calibration PA0-PA0, x1 */ [0b100011] = AVR_ADC_DIFF (0, 0, 20), /* ofs. calibration */
+            [0b001000] = AVR_ADC_DIFF (0, 1, 1), [0b001001] = AVR_ADC_DIFF (0, 1, 20),
+            [0b001010] = AVR_ADC_DIFF (0, 3, 1), [0b001011] = AVR_ADC_DIFF (0, 3, 20),
+            [0b101000] = AVR_ADC_DIFF (1, 0, 1), [0b101001] = AVR_ADC_DIFF (1, 0, 20),
+            [0b001100] = AVR_ADC_DIFF (1, 2, 1), [0b001101] = AVR_ADC_DIFF (1, 2, 20),
+            [0b001110] = AVR_ADC_DIFF (1, 3, 1), [0b001111] = AVR_ADC_DIFF (1, 3, 20),
+            [0b101100] = AVR_ADC_DIFF (2, 1, 1), [0b101101] = AVR_ADC_DIFF (2, 1, 20),
+            [0b010000] = AVR_ADC_DIFF (2, 3, 1), [0b010001] = AVR_ADC_DIFF (2, 3, 20),
+            [0b101010] = AVR_ADC_DIFF (3, 0, 1), [0b101011] = AVR_ADC_DIFF (3, 0, 20),
+            [0b101110] = AVR_ADC_DIFF (3, 1, 1), [0b101111] = AVR_ADC_DIFF (3, 1, 20),
+            [0b110000] = AVR_ADC_DIFF (3, 2, 1), [0b110001] = AVR_ADC_DIFF (3, 2, 20),
+            [0b100100] = AVR_ADC_DIFF (3, 3, 1), [0b100101] = AVR_ADC_DIFF (3, 3, 20), /* ofs. calibration */
+            [0b010010] = AVR_ADC_DIFF (3, 4, 1), [0b010011] = AVR_ADC_DIFF (3, 4, 20),
+            [0b010100] = AVR_ADC_DIFF (3, 5, 1), [0b010101] = AVR_ADC_DIFF (3, 5, 20),
+            [0b010110] = AVR_ADC_DIFF (3, 6, 1), [0b010111] = AVR_ADC_DIFF (3, 6, 20),
+            [0b011000] = AVR_ADC_DIFF (3, 7, 1), [0b011001] = AVR_ADC_DIFF (3, 7, 20),
+            [0b110010] = AVR_ADC_DIFF (4, 3, 1), [0b110011] = AVR_ADC_DIFF (4, 3, 20),
+            [0b011010] = AVR_ADC_DIFF (4, 5, 1), [0b011011] = AVR_ADC_DIFF (4, 5, 20),
+            [0b110100] = AVR_ADC_DIFF (5, 3, 1), [0b110101] = AVR_ADC_DIFF (5, 3, 20),
+            [0b111010] = AVR_ADC_DIFF (5, 4, 1), [0b111011] = AVR_ADC_DIFF (5, 4, 20),
+            [0b011100] = AVR_ADC_DIFF (5, 6, 1), [0b011101] = AVR_ADC_DIFF (5, 6, 20),
+            [0b110110] = AVR_ADC_DIFF (6, 3, 1), [0b110111] = AVR_ADC_DIFF (6, 3, 20),
+            [0b111100] = AVR_ADC_DIFF (6, 5, 1), [0b111101] = AVR_ADC_DIFF (6, 5, 20),
+            [0b011110] = AVR_ADC_DIFF (6, 7, 1), [0b011111] = AVR_ADC_DIFF (6, 7, 20),
+            [0b111000] = AVR_ADC_DIFF (7, 3, 1), [0b111001] = AVR_ADC_DIFF (7, 3, 20),
+            [0b111110] = AVR_ADC_DIFF (7, 6, 1), [0b111111] = AVR_ADC_DIFF (7, 6, 20),
+            [0b100110] = AVR_ADC_DIFF (7, 7, 1), [0b100111] = AVR_ADC_DIFF (7, 7, 20), /* ofs. calibration */
+
+            [32] = AVR_ADC_REF(0),       // 0V AGND
+            [33] = AVR_ADC_REF(1100),    // 1.1V internal Vref
+            [34] = AVR_ADC_TEMP(),
         },
 
         .adc = {

```

Thanks

edgargrimberg, 2019-04-16:
why not a PR?

djfd, 2019-04-17:
@edgargrimberg 
Because never did it/have no time to learn the procedure. Sorry.

JarrettBillingsley, 2019-05-05:
I really wish GH would let you turn a patch *into* a PR without having to fork, pull, apply patch, push, blah blah blah blaaaaaaah

gatk555, 2021-03-22:
... a patch into a PR ...
PR 439.  Trying to learn git, and beginning to like it.

gatk555, 2021-03-22:
Nice patch BTW.  References to datasheet pages? Whatever next!

djfd, 2021-03-22:
@gatk555  Thank  you so much, my friend!

What do you mean asking for " References to datasheet pages?", BTW? There is a couple of references for both Attiny24/24A at the very first post, is it not enough? Just checked, they are still working.

Thanks again for your efforts!

gatk555, 2021-03-24:
You are welcome!  It seems a nifty bit of software, but a pity that it is
now maintained so little.

I do not understand your comment beginning "What do you mean by asking
...", but I think it does not refer to anything I did or wrote.  All I did
was
apply the patch, build, commit, push, and then go to the site and make a
pull request.  It was the quoted page numbers in your patch that made me
think I was looking at something good, as there seems to be some rubbish
among the orphaned suggested changes.

gatk555, 2021-03-24:
But looking again here, I see that I did say that.  It was intended as approval of their inclusion in the patch.  Further proof that irony does not work on the internet, if any was needed.

gatk555, 2021-04-04:
The patch is now integrated, so this issue should be closed.

closed 2024-11-08 by gatk555