buserror/simavr #345: Consider providing a step by step guide to running an Arduino sketch using simavr

TomasHubelbauer, 2019-08-22:
Hi there, this project looks like exactly what I need from my searching around, but I am not capable of figuring out how to actually use it for my usecase. I have an Arduino sketch which computes some stuff (based on data included in its contents) and writes the results to the serial. I would like to run this sketch using simavr and see the results written to serial in my terminal. I read the readme, the manual (way over my head) and searched through GitHub issues containing *arduino*, which led me to #270 where I saw @idoc had the same usecase, so I also asked them for the same in that (closed) issue. If they reply, I will PR the project README with the guide after I verify I was able to use it.

lcgamboa, 2019-08-22:
1- Use the Arduino or avr-gcc to compile the sketch to generate the .hex file.
2- Use the patched version of simduino of my fork https://github.com/lcgamboa/simavr/ with option SIMAVR_UART_XTERM=1 and the .hex file.

Obs: avrdude is for transfer the .hex file over serial to arduino bootloader, this is not necessary in your usecase.

You can try my graphic arduino simulator https://github.com/lcgamboa/picsimlab as a "virtual arduino uno" with the Arduino IDE.

TomasHubelbauer, 2019-08-22:
I tried to use the Arduino CLI to compile the sketch (I don't like the Arduino IDE very much so I try to see if I can do this part using CLI tools only) and it spit out an ELF file. Is that equivalent to a HEX file or does it need further processing?

SimAVR says it can load ELF files. Could you please provide the exact command for your bullet point no. 2? I don't know how to construct it. So even if HEX and ELF are different, I should already be able to use the ELF with SimAVR once I know what command to use exactly.

Does PicSimLab support ELF? It looks really good and the manual looks great! I am on macOS right now so I will test it out on Windows tomorrow.

TomasHubelbauer, 2019-08-22:
Nevermind, sorry, it spit out both an ELF and a HEX file. I missed it at first 🙄

TomasHubelbauer, 2019-08-23:
Closed by mistake. Reopening. I didn't find other issues dealing with either of the two errors, so I'm not sure how to proceed to try and fix either of them.

TomasHubelbauer, 2019-09-14:
My bad! With the latest version of SimAVR when running the last line in the above script, `run_avr` actually correctly outputted the expected serial values… I just mistook them for SimAVR's own logging… 🙄 Really dumb of me, but now that I realized that, I can say this project works perfectly! Thank you!

closed 2019-09-14 by TomasHubelbauer