buserror/simavr #350: Running a CALL opcode results in half of the PC value being pushed to the stack

progirep, 2019-11-22:
It seems as if running a CALL opcode in sinavr (when simulating on an ATMega2560 platform) results in half of the address of the next instruction to be executed after the CALL instruction is pushed to the stack.

However, the real ATMega2560 seems to push the actual address of the next instruction (to be executed after the subroutine called) to the stack.

This means that the behavior of the simulator deviates from the real AVR ICs. I can provide a more concrete example if needed.

bsekisser, 2020-10-11:
in sim_core.c, line ~1225:

`
						case 0x940e:
						case 0x940f: {	// CALL -- Long Call to sub, 32 bits -- 1001 010a aaaa 111a
							avr_flashaddr_t a = ((opcode & 0x01f0) >> 3) | (opcode & 1);
							uint16_t x = _avr_flash_read16le(avr, new_pc);
							a = (a << 16) | x;
							STATE("call 0x%06x\n", a);
//							new_pc += 2;
							cycle += 1 + _avr_push_addr(avr, new_pc);
							new_pc = a << 1;
							TRACE_JUMP();
							STACK_FRAME_PUSH();
						}	break;
`

Take out the line before the push: "new_pc +=2" and see if that fixes it...  that looks to be incorrect.

edgar-bonet, 2020-10-11:
@progirep: I think a testable example should be useful.

@bsekisser: That line looks correct to me. `new_pc` is initialized to point to the word just after the first word of the instruction being currently executed. In other words, it's the already-incremented PC. If this turns out to be a 32-bit instruction, `new_pc` has to be incremented by an extra word. And indeed, the line

```c
new_pc += 2;
```

is also found in the implementation of the other 32-bit instructions: `lds` and `sts`. Not in `call` because that instruction discards the value of the incremented PC.

bsekisser, 2020-10-11:
Whoa!  Blind moment!  You are absolutely correct!  Missed the flash read.  Needed sleep!

progirep, 2020-10-12:
Ok, so I promised a more concrete example.

For instance, I compiled the following Assembly code using `avr-gcc klausurBeispiel.S -mmcu=atmega2560 -Os -g -o klausurBeispiel.elf`:

```
#include <avr/io.h>

.global main

main:

    subi r24, 0xFF
    eor r0, r0
    out 0x3f, r0
    ldi r24, 0x10
    call 0x200
    subi r24, 0xFD
    subi r24, 0xFF

    SEI
    LDI R18, 0x02
    LDI R19, 0x7F
    ADD R18, R19
    ADC R18, R19
    LDI R20, 250
    ADD R18, R20
    ADC R18, R19
    INC R19
    LDI R20, 34
    ADD R19, R20
        
    rjmp main

```

It doesn't rally matter that this code doesn't actually do anything useful. I've written a script that plots an execution of simavr into LaTeX and compiled it to a PDF, so that the execution can be looked at step-by-step. There does not appear to be the possibility to attach a PDF to the issue, so I've put it here: [https://www.ruediger-ehlers.de/data/AVR-CALL-Example-For-SimAVR.pdf](https://www.ruediger-ehlers.de/data/AVR-CALL-Example-For-SimAVR.pdf) for you. In the memory dump on the left-hand side of the slides, it can be seen that when a function is called, the return address divided by two is pushed to the stack.

edgar-bonet, 2020-10-12:
This is the normal behavior.

At the hardware level, the AVR uses _word addresses_ to address the flash (save on `lpm`). Each address represents a 16-bit word within the flash memory. Many development tools, however, use instead _byte addresses_. This is the case, for example, of the GNU binutils (e.g. avr-objdump). The difference between byte addresses and word addresses is just a factor two:

| word address | byte addresses |
|--------------|----------------|
|    0x0000    | 0x0000 0x0001  |
|    0x0001    | 0x0002 0x0003  |
|    0x0002    | 0x0004 0x0005  |
|     ...      |      ...       |

The tool you are using displays the contents of the flash like this:

```text
⇒ 000000  0c 94 72 00    jmp 0xe4
  000004  0c 94 7e 00    jmp 0xfc
  000008  0c 94 7e 00    jmp 0xfc
  ...
```

You can see that the address is incremented by 4&nbsp;units for every 4&nbsp;bytes. It follows that this tool uses _byte addresses_, just like avr-objdump. The AVR hardware, however, stores a word address in the program counter, and that is what gets pushed on the stack on `call`.

You can safely close this issue as “invalid”.

progirep, 2020-10-12:
Well, the thing is that the memory dump of a real AVR ATMega2560 shows that at least on the stack, the addresses are not stored divided by two - I had this checked by an employee.

edgar-bonet, 2020-10-12:
This is highly surprising. Could you share a test code that shows different behavior on the simulation and the real hardware?

edgar-bonet, 2020-10-12:
OK, I wrote a test case myself. The following program issues `call .` in order to push the PC to the stack without changing the program flow. It then pops the address that has just been pushed and prints it to the serial port.

```lisp
#define BAUD 9600
#define F_CPU 16000000
#include <util/setbaud.h>
#include <avr/io.h>
#define io(x) _SFR_IO_ADDR(x)
#define mem(x) _SFR_MEM_ADDR(x)

init:
    ; Set the stack pointer to 0x01ff
    ldi   r28, lo8(0x01ff)
    ldi   r29, hi8(0x01ff)
    out   io(SPL), r28
    out   io(SPH), r29

    ; Test to see what CALL pushes.
    call  .  ; push the address of the following NOP to the stack
    nop      ; word address: 0x06, byte address: 0x0c

    ; Set up the UART.
    ldi   r16, UBRRL_VALUE
    sts   mem(UBRR0L), r16
    ldi   r16, UBRRH_VALUE
    sts   mem(UBRR0H), r16
    ldi   r16, USE_2X << U2X0
    sts   mem(UCSR0A), r16
    ldi   r16, _BV(TXEN0)
    sts   mem(UCSR0B), r16

    ; Print out the address that was pushed to the stack.
#if __AVR_3_BYTE_PC__
    pop   r16
    rcall print_hex
#endif
    pop   r16
    rcall print_hex
    pop   r16
    rcall print_hex
    ldi   r16, '\r'
    rcall print_byte
    ldi   r16, '\n'
    rcall print_byte
    sleep    ; stop the simulation
1:  rjmp 1b  ; stop the program on the hardware

    ; Printing routines.
print_hex:  ; print r16 in hex
    mov   r18, r16
    swap  r16
    rcall print_nibble
    mov   r16, r18
    rcall print_nibble
    ret

print_nibble:  ; print the low nibble of r16 in hex
    andi  r16, 0x0f
    cpi   r16, 10
    brcs  1f
    subi  r16, '9' + 1 - 'a'
1:  subi  r16, -'0'
print_byte:    ; output the byte r16
    lds   r17, mem(UCSR0A)
    andi  r17, _BV(UDRE0)
    breq  print_byte
    sts   mem(UDR0), r16
    ret
```

I stored it as “t1.S”, then tested it on both simavr and an Arduino Uno with the following commands:

```shell
# Compile.
avr-gcc -mmcu=atmega328p -nostdlib t1.S -o t1.elf

# Test with simavr.
run_avr -f 16000000 -m atmega328p t1.elf

# Test on the hardware.
avrdude -q -p atmega328p -c arduino -P /dev/ttyACM0 -b 115200 -U t1.elf
stty -F /dev/ttyACM0 raw 9600 && cat /dev/ttyACM0
```

Both simavr and the real hardware print `0006`, which is the _word address_ of the instruction following the `call`. As this happens to match the documented behavior, it looks to me like you just made a mistake somewhere.

progirep, 2020-10-12:
Thanks for checking this! As mentioned, I had this checked by an employee, but if you just verified it on the ATMega328, then it's probably correct as it is in the simulator.

closed 2020-10-12 by progirep