buserror/simavr #362: Exit simAVR with a non-zero exit code, triggered by the atMega program (for unittests)

cdwijs, 2020-01-20:
Hi All,
 
I would like to exit simAVR with a non-zero exit code, for instance if a byte in the atMega is non-zero.  This would allow me to remotely run unit tests in a simulated atMega32, and have failed reports be reported back to me. Maybe this can be done by defining a new simAVR command like  SIMAVR_CMD_TERMINATE_NONZERO.

Here's the workflow I'm trying to setup. Step 9 does not work, as simAVR always terminates with a zero exit code.

1) On a windows PC, I write a source code in C for my atMega32 microcontroller.
2) I push the source code to a Gitlab server
3) The gitlab server parses a special file in the root of the source code (.gitlab-ci.yml), starts a docker container, and runs the test script in the docker container.
4) In the docker container, the test script compiles the source code into an atMega32 program.
5) In the docker container, the test script runs simAVR. simAVR runs the atMega32 program.
6) In simAVR, the atMega program runs a series of unit tests, and reports to the serial port. simAVR forwards the data from the atMega program to it's standard output.
7) When the atMega program is finished, it calls sleep_cpu(); This causes simAVR also to close.
8) the test script also ends, the output from the test script is send back to the Gitlab server, and the docker container is destroyed.
9) If the test script ends with a non-zero exit code, the Gitlab server considers the test as failed, and sends an e-mail to me with the output from the test script.

I'm using the current version of simavr, commit ab4c17c
https://github.com/buserror/simavr/tree/ab4c17cad37f33e956d625178a9904fc7c3056c2

Cheers,
Cedric de Wijs

cdwijs, 2020-01-21:
I've implemented 2 extra commands, one that ends run_avr with an exit code of 0, and one that ends run_avr with an exit code of 1. The code is crude, but it is sufficient for my use case.

I've submitted this pull request with my code:
https://github.com/buserror/simavr/pull/363

edgar-bonet, 2020-06-06:
Note that Gitlab cares about the exit code of the test script, not the one of simavr. The test script could parse the output of the ATmega32 program (forwarded by simavr) and set its own exit code accordingly.

gatk555, 2025-12-07:
The topic need not be narrowly defined.  As @edgar-bonet noted, with minimal scripting, no change is needed.  If there is to be one, it should be a clear improvement.