buserror/simavr #367: simaver crashes: Aborted (core dumped) after loading .text and .data from a.out file.

cdwijs, 2020-03-09:
Hi All,

I've managed to crash simavr. Steps I've taken:
1) Compile an atmega32 program into an a.out file.
2) Compile simavr from this push request: https://github.com/buserror/simavr/pull/363
3) Use runavr to run the a.out file. This yields the following console output:
```
$ ./simavr/simavr/run_avr 3736894-temperatuur-regelaar/a.out 
Loaded 36232 .text at address 0x0
Loaded 472 .data
Aborted (core dumped)
```

My versions:
Simavr: https://github.com/buserror/simavr/pull/363
$ uname -a
Linux cedric-Ubuntu1804 5.3.0-26-generic #28~18.04.1-Ubuntu SMP Wed Dec 18 16:40:14 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

strace output:
`
$ strace ./simavr/simavr/run_avr 3736894-temperatuur-regelaar/a.out 
execve("./simavr/simavr/run_avr", ["./simavr/simavr/run_avr", "3736894-temperatuur-regelaar/a.o"...], 0x7ffd8e66f9b8 /* 60 vars */) = 0
brk(NULL)                               = 0x559c44703000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/tls", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/x86_64", 0x7ffcdf697080) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/libsimavr.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320q\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1848504, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fca1933e000
mmap(NULL, 3020976, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fca18e37000
mprotect(0x7fca18e5b000, 2093056, PROT_NONE) = 0
mmap(0x7fca1905a000, 782336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x23000) = 0x7fca1905a000
close(3)                                = 0
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/libelf.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=188286, ...}) = 0
mmap(NULL, 188286, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fca19310000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libelf.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0.\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=104952, ...}) = 0
mmap(NULL, 2199936, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fca18c1d000
mprotect(0x7fca18c36000, 2093056, PROT_NONE) = 0
mmap(0x7fca18e35000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7fca18e35000
close(3)                                = 0
openat(AT_FDCWD, "/home/crldewijs/simavr/simavr/../simavr/obj-x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260\34\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2030544, ...}) = 0
mmap(NULL, 4131552, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fca1882c000
mprotect(0x7fca18a13000, 2097152, PROT_NONE) = 0
mmap(0x7fca18c13000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7fca18c13000
mmap(0x7fca18c19000, 15072, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fca18c19000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220\37\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=116960, ...}) = 0
mmap(NULL, 2212016, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fca1860f000
mprotect(0x7fca1862b000, 2093056, PROT_NONE) = 0
mmap(0x7fca1882a000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b000) = 0x7fca1882a000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fca1930e000
arch_prctl(ARCH_SET_FS, 0x7fca1930eb80) = 0
mprotect(0x7fca18c13000, 16384, PROT_READ) = 0
mprotect(0x7fca1882a000, 4096, PROT_READ) = 0
mprotect(0x7fca18e35000, 4096, PROT_READ) = 0
mprotect(0x7fca1905a000, 774144, PROT_READ) = 0
mprotect(0x559c4425b000, 4096, PROT_READ) = 0
mprotect(0x7fca19340000, 4096, PROT_READ) = 0
munmap(0x7fca19310000, 188286)          = 0
openat(AT_FDCWD, "3736894-temperatuur-regelaar/a.out", O_RDONLY) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\2\0S\0\1\0\0\0\0\0\0\0004\0\0\0"..., 52) = 52
fcntl(3, F_GETFD)                       = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=51588, ...}) = 0
pread64(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\2\0S\0\1\0\0\0\0\0\0\0004\0\0\0"..., 64, 0) = 64
brk(NULL)                               = 0x559c44703000
brk(0x559c44724000)                     = 0x559c44724000
pread64(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 560, 51028) = 560
pread64(3, "\0.symtab\0.strtab\0.shstrtab\0.mmcu"..., 133, 50894) = 133
pread64(3, "\1@atmega32\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 152, 36884) = 152
pread64(3, "\f\224E\1\f\224b\1\f\224b\1\f\224b\1\f\224b\1\f\224b\1\f\224b\1\f\224b\1"..., 36232, 180) = 36232
pread64(3, "\0\0\0\2\0\0\0\0\307\1\0\0\0\0Starting Unittests"..., 472, 36412) = 472
pread64(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\221\0\0\0\0\0\3\0\1\0"..., 6640, 39992) = 6640
pread64(3, "\0main.c\0__SP_H__\0__SP_L__\0__SREG"..., 4262, 46632) = 4262
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 4), ...}) = 0
write(1, "Loaded 36232 .text at address 0x"..., 34Loaded 36232 .text at address 0x0
) = 34
write(1, "Loaded 472 .data\n", 17Loaded 472 .data
)      = 17
close(3)                                = 0
brk(0x559c44746000)                     = 0x559c44746000
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [], 8) = 0
getpid()                                = 12093
gettid()                                = 12093
tgkill(12093, 12093, SIGABRT)           = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=12093, si_uid=113133} ---
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)
`

cdwijs, 2020-03-09:
Ive run simavr in gdb, and created a backtrace:

```
$ gdb --args ./simavr/simavr/run_avr 3736894-temperatuur-regelaar/a.out
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./simavr/simavr/run_avr...done.
(gdb) r
Starting program: /home/crldewijs/simavr/simavr/run_avr 3736894-temperatuur-regelaar/a.out
Loaded 36232 .text at address 0x0
Loaded 472 .data

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7528801 in __GI_abort () at abort.c:79
#2  0x00007ffff7b02fa0 in avr_loadcode (avr=<optimized out>, code=<optimized out>, size=<optimized out>, address=<optimized out>) at sim/sim_avr.c:251
#3  0x00007ffff7aff8cd in avr_load_firmware (avr=0x55555575b240, firmware=0x7fffffffd2f0) at sim/sim_elf.c:76
#4  0x000055555555511f in main (argc=<optimized out>, argv=0x7fffffffddf8) at sim/run_avr.c:261
```

cdwijs, 2020-03-09:
In the source of simavr I see my program is too big:
```
sim_avr.c, line 248-252:
if ((address + size) > avr->flashend+1) {
		AVR_LOG(avr, LOG_ERROR, "avr_loadcode(): Attempted to load code of size %d but flash size is only %d.\n",
			size, avr->flashend + 1);
		abort();
	}
```
What I don't understand is why I don't see the message about the code size in the console. Where does simavr log to?

buserror, 2020-03-29:
"file a.out" it's likely it's not a valid AVR ELF file

cdwijs, 2020-03-30:
You are correct that the AVR ELF file is way too big. I would however have liked a more human readable error message.

vintagepc, 2020-10-08:
> In the source of simavr I see my program is too big:
>
> ```
> sim_avr.c, line 248-252:
> if ((address + size) > avr->flashend+1) {
> 		AVR_LOG(avr, LOG_ERROR, "avr_loadcode(): Attempted to load code of size [...]

AVR_LOG is controlled by the number of -v arguments. Run with `-vvv`  for example, and you should see it.

gatk555, 2025-04-07:
The message is now shown by default. Change d3403210ba.