buserror/simavr #371: GDB Error: Remote target does not support flash erase

valeros, 2020-05-08:
Hi!
The latest `simavr` from the master branch is not able to execute `load` command from GDB:

`
target remote :1234
`

msquirogac, 2020-05-30:
@buserror
To fix this properly, support for the following gdb packets must be added
1) vFlashErase
1) vFlashWrite
1) vFlashDone

I already got [support](https://github.com/msquirogac/simavr/tree/fix-371) for the first and last, yet the second one is a bit more challenging.

As [documentation](https://sourceware.org/gdb/current/onlinedocs/gdb/Packets.html) says, vFlashWrite uses the binary encoding mode like the `X` packet instead of the hex mode that most of the packets use.
1) Do you have already have a decoder for said mode? No problem if you don't, just wanted to be sure to avoid repeating code.
2) Unlike the `M` packet, `vFlashWrite` one doesn't have a length field and was thinking to add it as an argument of the `gdb_handle_command` function. Any problem with that?

buserror, 2020-08-31:
@msquirogac I think there is support for something like that in the self programing code -- ie the stuff the bootloader uses? not sure tho. Otherwise I coul dhave a look. And yes, as @enekochan mention, if you have a PR with what you already have... thanks!

OzPineapple, 2022-03-10:
I have the same problem using the package from Arch repo.
```
avr-gcc -mmcu=atmega8 -g -o test.elf test.S
simavr -m atmega8 -f 8000000 -g -v test.elf
avr-gdb test.elf -ex "target remote :1234" -ex "load"
```
So I get:
```
Reading symbols from test.elf...
Remote debugging using :1234
0x00000000 in __vectors ()
Remote target does not support flash erase
```
I do not understand if I need to do some extra config setup to make it work. May this also makes that the `display` comand shows nothing?

This the test.S file btw:
```
#include <avr/io.h>

.global main

main:
	ldi r16, 0x01
	ldi r17, 0x02
	add r16, r17
	ldi r17, 0xFF
```

gatk555, 2022-03-13:
I found some time to try this yesterday and seem to have a working implementation of the three commands.  But I do not understand what the use of it is.  In the example above, "-ex load" is not necessary, as simavr has already loaded the file.
I suppose it could be convenient to re-compile and continue the same gdb session, but found that simavr enters an infinite loop after reset.

PR #482 (2022-04-11): Gdb: add support for "load" and "info_registers" and fix bugs.
Addition of load command fixes #371.  The "awatch" command now works and "break" instructions are useable without having to make gdb jump over them by hand.  General gdb behaviour is improved by not sending confusing stop reports.  Other bug fixes as described in the commit comments, including an equivalent to the fix in #480.
closed 2022-04-11 by buserror