buserror/simavr #375: Update display_usage function

msquirogac, 2020-05-25:
There are some arguments that already work, but aren't listed or explained. Said arguments are the following ones:
`--vcd-trace-file`
`--vcd-trace-name`
`--add-vcd-trace`

msquirogac, 2020-05-25:
I found some redundant code
~~~
		} else if (!strcmp(argv[pi], "--vcd-trace-name")) {
			if (pi + 1 >= argc) {
				fprintf(stderr, "%s: missing mandatory argument for %s.\n", argv[0], argv[pi]);
				exit(1);
			}
			++pi;
			snprintf(f.tracename, sizeof(f.tracename), "%s",  argv[pi]);
~~~

~~~
		} else if (!strcmp(argv[pi], "--vcd-trace-file")) {
			if (pi + 1 >= argc) {
				fprintf(stderr, "%s: missing mandatory argument for %s.\n", argv[0], argv[pi]);
				exit(1);
			}
			snprintf(f.tracename, sizeof(f.tracename), "%s", argv[++pi]);
		}
~~~

As both **--vcd-trace-file** and **--vcd-trace-name** do the same, they should merge into one.
Maybe the name should change to **--output** and **-o** to make it like the --input argument.

buserror, 2020-05-27:
Feel free to PR me that patch, it makes sense...!

closed 2020-06-02 by msquirogac