buserror/simavr #382: register_notify for entry Port

ricardojlrufino, 2020-06-09:
is there any way to optimize the code below? instead of registering pin by pin, monitor all?
```
	// connect all the pins on port B to our callback
	for (int i = 0; i < 8; i++){
		avr_irq_register_notify(
			avr_io_getirq(avr, AVR_IOCTL_IOPORT_GETIRQ('B'), i),pins_portB_hook, NULL
		);

		avr_irq_register_notify(
			avr_io_getirq(avr, AVR_IOCTL_IOPORT_GETIRQ('C'), i),pins_portC_hook, NULL
		);

		avr_irq_register_notify(
			avr_io_getirq(avr, AVR_IOCTL_IOPORT_GETIRQ('D'), i),pins_portD_hook, NULL
		);
	}
```

buserror, 2020-06-09:
Sure, therie is a IOPORT_ALL in ioport.h, that will send you a byte with all pins for that port...

ricardojlrufino, 2020-06-09:
IOPORT_IRQ_PIN_ALL  ?!

buserror, 2020-06-09:
Yarp!