buserror/simavr #383: ADC Channels > 7 not working.

Arcachofo, 2020-06-18:
Looks like they are not implemented...

simavr/sim/avr_adc.h 138, struct avr_adc_t
`uint16_t		adc_values[8];	// current values on the ADCs`

simavr/sim/avr_adc.c, 276, avr_adc_irq_notify
`	case ADC_IRQ_ADC0 ... ADC_IRQ_ADC7: {`

But they are here:

simavr/sim/avr_adc.h 44
```
enum {
	// input IRQ values. Values are /always/ volts * 1000 (millivolts)
	ADC_IRQ_ADC0 = 0, ADC_IRQ_ADC1, ADC_IRQ_ADC2, ADC_IRQ_ADC3,
	ADC_IRQ_ADC4, ADC_IRQ_ADC5, ADC_IRQ_ADC6, ADC_IRQ_ADC7,
	ADC_IRQ_ADC8, ADC_IRQ_ADC9, ADC_IRQ_ADC10, ADC_IRQ_ADC11,
	ADC_IRQ_ADC12, ADC_IRQ_ADC13, ADC_IRQ_ADC14, ADC_IRQ_ADC15,
	ADC_IRQ_TEMP,			// see the datasheet
	ADC_IRQ_IN_TRIGGER,
	ADC_IRQ_OUT_TRIGGER,	// sends a avr_adc_mux_t
	ADC_IRQ_COUNT
};
```

buserror, 2020-06-30:
Looks like a nice first pull request for someone! :-) :-)

vintagepc, 2020-07-23:
Heh, I've already fixed this for something I was working on. That branch has some tweaks that are probably not wanted upstream though. 

I'll see if I can isolate the beneficial changes and file a PR.

PR #389 (2020-08-03): Fix ADCs>7
Fix #383 - Enable read IRQs for ADCs > 7
closed 2020-08-03 by buserror