buserror/simavr #386: Compilation fails with gcc 8.3.0

sillydan1, 2020-07-03:
When compiling with gcc 8.3.0 There's a warning on `run_avr.c`. 

```CC sim/run_avr.c
sim/run_avr.c: In function ‘main’:
sim/run_avr.c:109:5: error: ‘strncpy’ specified bound 24 equals destination size [-Werror=stringop-truncation]
     strncpy(name, argv[++pi], sizeof(name));
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

The fairly simple solution is to just ignore the warning in this particular instance, since it looks like it is correct usage of `strncpy`

```C
if (pi < argc-1) {
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wstringop-truncation"
    strncpy(name, argv[++pi], sizeof(name));
#pragma GCC diagnostic pop
}
```

edgar-bonet, 2020-07-03:
On the one hand, this warning is legit, and should not be dismissed like this. That line of code can create a string that is not NUL-terminated.

On the other hand, this issue has already been reported (#293) and fixed (commit c2c8e40ff219f78779fc2743f0c1ffbabe159e37, pull request #294).

sillydan1, 2020-07-06:
It would be nice with a release soon then. The "newest" release is from 2018. Any idea of when 1.7 (or maybe just a patch) will get tagged as a release?

closed 2020-07-24 by sillydan1