buserror/simavr #400: Added traces are cleared if elf file is passed later on command line

couchand, 2020-09-17:
Passing the input elf file on the command line after `--add-trace` parameters causes the added traces to be silently dropped.  This is pretty surprising, and caused me much headache.

It looks like the argument processing for `--add-trace` directly modifies the same `elf_firmware_t` that is passed to `elf_read_firmware`.  Two possible mitigations would be to accumulate the trace structs separately and later merge, or update `elf_read_firmware` to not clobber traces already added.  I couldn't immediately find where the latter was happening.

If you'd like to suggest the right strategy to take and point me in the right direction, I'd be happy to submit a patch.  Thanks!

gatk555, 2021-03-28:
It sounds similar to the cause in Pull Request #438: elf_read_firmware() starts with memset(), destroying all initialiasations.  Removing memset() crashes, so a fix may not be so simple ...

gatk555, 2021-03-28:
It sounds similar to the cause in Pull Request #438: elf_read_firmware() starts with memset(), destroying all initialiasations.  Removing memset() crashes in tests, so a fix may not be so simple ...

buserror, 2021-04-01:
We might need a split functionality for the firmware loading approach then, with an _init that prepares the structure, leave user code a way to add their own things, and then do an read_firmware() to finish it up.

gatk555, 2025-03-31:
Believed fixed by PR #440.

closed 2025-03-31 by gatk555